autowarefoundation/autoware_universe #1468: Handling bus stops

BonoloAWF, 2022-07-29:
### Checklist

- [X] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [X] I've searched other issues and no duplicate issues were found.
- [X] I've agreed with the maintainers that I can plan this task.

### Description

As part of the Bus ODD, the vehicle is required to approach a bus stop, stop within the designated area whilst passengers embark/disembark, and then continue along the route. 

This functionality needs to be added to Autoware.Universe. This issue is to track the progress of this addition. 

### Purpose

To facilitate bus stop behaviour

### Possible approaches

Possible approaches are summarised in this discussion: https://github.com/orgs/autowarefoundation/discussions/2651
- [ ] Updating the planner to handle start/stop behaviour
- [ ] Scheduling bus stop start/stop behaviour using the existing web tool(s)

### Definition of done

The bus can handle bus stops

BonoloAWF, 2022-08-10:
cc: @mitsudome-r @MertClk @hueirutseng @xmfcx @HaoruXue 
Meeting minutes from AWF/ITIR/LeoDrive meeting held on 10 August 2022:
1. Timelines for bus ODD
- ITRI bus will be tested on campus in September, post vehicle integration 
- LeoDrive internal project final tests and demo are in September. The LeoDrive bus will be available for the AWF demo until 31 December 2022
- AWF - the bus ODD demos will start in early December 

2. Handling bus stops
- Route scheduling and bus stops will be handled on the server side - each bus stop will have a unique pose associated with it, each new goal is set via this external service (no modifications to the planner)
- Ryohsuke will investigate whether the existing TierIV tool can be used for the demo
- If the TierIV tool is not available for use, a similar tool will be implemented by AWF
- The bus will stop at every bus stop along the route 
- 
4. Handling passenger embarking/disembarking
5. Bus user interfaces

MertClk, 2022-08-10:
**ISUZU Bus Displays**

Below, I'm sharing displays and panel existing on our bus. We can use them to inform environment and passengers. 

1. Front Destination Display  
[image]

2. Right Side Destination Display
[image]

3. Interior Destination Display 
[image]

4. Driver Control Panel
[image]

hueirutseng, 2022-08-11:
ITRI Bus Display 
1. Front display will show "Autonomous Bus" 
    
[image]

2. Interior display for passengers

[image]

3. Interior display for test driver

[image]

You might also refer to a demo video showing the idea that ITRI autonomous bus operated on public roads.
[https://drive.google.com/file/d/1ngEqr8b8FmQi4Xb1Ri4C52ffHogG2Slf/view?usp=sharing]

BonoloAWF, 2022-08-17:
All information has been provided. Closing.

closed 2022-08-17 by BonoloAWF