autowarefoundation/autoware_universe #1476: behavior path planner better verbose

taikitanaka3, 2022-08-01:
### Checklist

- [X] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [X] I've searched other issues and no duplicate issues were found.
- [X] I've agreed with the maintainers that I can plan this task.

### Description

for the most of cases behavior path planner has zero shift at departure and return shift point 
so this warning is happen all the time.
```
] [planning.scenario_planning.lane_driving.behavior_planning.behavior_path_planner.Avoidance]: Direction is UNKNOWN, distance = -1.79769e+308
[component_container_mt-13] [WARN] [1659327082.092174501] [planning.scenario_planning.lane_driving.behavior_planning.behavior_path_planner.Avoidance]: Direction is UNKNOWN, distance = -1.79769e+308
[component_container_mt-13] [WARN] [1659327082.096028470] [planning.scenario_planning.lane_driving.behavior_planning.behavior_velocity_planner.intersection_module]: setStopLineIdx fail
[component_container_mt-13] [WARN] [1659327082.205619825] [planning.scenario_planning.lane_driving.behavior_planning.behavior_path_planner.Avoidance]: Direction is UNKNOWN, distance = -1.79769e+308
[component_container_mt-13] [WARN] [1659327082.292172711] [planning.scenario_planning.lane_driving.behavior_planning.behavior_path_planner.Avoidance]: Direction is UNKNOWN, distance = -1.79769e+308
[component_container_mt-13] [WARN] [1659327082.392357136] [planning.scenario_planning.lane_driving.behavior_planning.behavior_path_planner.Avoidance]: Direction is UNKNOWN, distance = -1.79769e+308
[component_container_mt-13] [WARN] [1659327082.493445367] [planning.scenario_planning.lane_driving.behavior_planning.behavior_path_planner.Avoidance]: Direction is UNKNOWN, distance = -1.79769e+308
[component_container_mt-13] [WARN] [1659327082.592359287] [planning.scenario_planning.lane_driving.behavior_planning.behavior_path_planner.Avoidance]: Direction is UNKNOWN, distance = -1.79769e+308
[component_container_mt-13] [WARN] [1659327082.692397839] [planning.scenario_planning.lane_driving.behavior_planning.behavior_path_planner.Avoidance]: Direction is UNKNOWN, distance = -1.79769e+308
[component_container_mt-13] [WARN] [1659327082.793275756] [planning.scenario_planning.lane_driving.behavior_planning.behavior_path_planner.Avoidance]: Direction is UNKNOWN, distance = -1.79769e+308
[component_container_mt-13] [WARN] [1659327082.892204832] [planning.scenario_planning.lane_driving.behavior_planning.behavior_path_planner.Avoidance]: Direction is UNKNOWN, distance = -1.79769e+308
[component_container_mt-13] [WARN] [1659327082.992716297] [planning.scenario_planning.lane_driving.behavior_planning.behavior_path_planner.Avoidance]: Direction is UNKNOWN, distance = -1.79769e+308
[component_container_mt-13] [WARN] [1659327083.103483370] [planning.scenario_planning.lane_driving.behavior_planning.behavior_path_planner.Avoidance]: Direction is UNKNOWN, distance = -1.79769e+308
[component_container_mt-13] [WARN] [1659327083.110933881] 
```

### Purpose

to reduce unnecessary warning

### Possible approaches

change to comment
```
    if (candidate.lateral_shift > 0.0) {
      rtc_interface_left_.updateCooperateStatus(
        uuid_left_, isExecutionReady(), candidate.distance_to_path_change, clock_->now());
      return;
    } else if (candidate.lateral_shift < 0.0) {
      rtc_interface_right_.updateCooperateStatus(
        uuid_right_, isExecutionReady(), candidate.distance_to_path_change, clock_->now());
      return;
    } else {
      // departure or return shift
      return;
    }
```

### Definition of done

no warning at nominal case
[image]

taikitanaka3, 2022-08-01:
The problem of this is CandidateOutput is created even though candidate lateral shift is zero shift

taikitanaka3, 2022-08-01:
@shulanbushangshu 
Thank you for PR I have confirmed this removes unnecessary candidateOutputs.
 However zero shift should not be one of the candidateOutputs. Do you know why this happends?
[image]

shulanbushangshu, 2022-08-01:
@taikitanaka3 .
zero shift becoming one of the candidateOutputs  seems unlikely for the reason：
1.In the planCandidate of AvoidanceModule, it just set value to the shifted point when the new_shift_points are found;
2. when there are no new_shift_points,zero is set to shift_point of planCandidate by default.

I think the original purpose is to prevent the calculation error of offset distance and add alarm information.

taikitanaka3, 2022-08-01:
@shulanbushangshu cc: @rej55 
Thankyou I think we should not make candidateOutput like lane change module.
because zero shift can introduce bugs

taikitanaka3, 2022-08-02:
@shulanbushangshu 
Thank you for fix PR but this error is caused due to 
clearWaitingApproval() when there is no new shift points and  then planWaitingApproval() is called and continue calling planWaitingApproval() even though there is no new shift points

@rej55 -san is going to fix this error with avoidance distance calculation so can you wait until he fix this issue with RTC?

taikitanaka3, 2022-08-05:
@shulanbushangshu 
@rej55 -san solved this issue with this PR can you re-open this issue if you have any farther problem?
https://github.com/autowarefoundation/autoware.universe/pull/1510

closed 2022-08-05 by taikitanaka3