autowarefoundation/autoware_universe #1487: Wrong use of min_velocity_for_map_based_prediction parameter

mitsudome-r, 2022-08-01:
### Checklist

- [X] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [X] I've searched other issues and no duplicate issues were found.
- [X] I'm convinced that this is not my fault but a bug.

### Description

In map_based_prediction package, there is a parameter called `min_velocity_for_map_based_prediction`. The description the [README](https://github.com/autowarefoundation/autoware.universe/blob/main/perception/map_based_prediction/README.md#parameters) states "apply map-based prediction to the objects with higher velocity than this value" so it is the threshold parameter to decide whether the target velocity is moving or not. It is actually used in that way for vehicle prediction. However, this is used as a minimum velocity for slowly moving pedestrians at crosswalk prediction. We should make sure that the parameter is used in consistent way.

Relevante Code:
* It is used as a threshold parameter to decide whether the vehicle is moving or not.
https://github.com/autowarefoundation/autoware.universe/blob/bc001184ece5ed0332c97f106be8f6d7a014311b/perception/map_based_prediction/src/map_based_prediction_node.cpp#L416

* For crosswalk, it is used as minimum velocity for prediction path for stopped targets to detect potentially crossing objects.
https://github.com/autowarefoundation/autoware.universe/blob/bc001184ece5ed0332c97f106be8f6d7a014311b/perception/map_based_prediction/src/map_based_prediction_node.cpp#L529-L531

### Expected behavior

`min_velocity_for_map_based_prediction` is used in consistent way

### Actual behavior

`min_velocity_for_map_based_prediction` is used differently for vehicle and pedestrians

### Steps to reproduce

1. Install Autoware and [Scenario Sim](https://autowarefoundation.github.io/autoware-documentation/main/tutorials/scenario-simulation/planning-simulation/scenario-test-simulation/)
2. Download and extract scenario file: [crosswalk_scenario.zip](https://github.com/autowarefoundation/autoware.universe/files/9235487/crosswalk_scenario.zip)
3. Modify the map path in the scenario file (L23 and L25)
4. Start rviz: `ros2 run rviz2 rviz2 -d $HOME/autoware/src/launcher/autoware_launch/autoware_launch/rviz/autoware.rviz`
5. In the different terminal run the following command:`ros2 launch scenario_test_runner scenario_test_runner.launch.py architecture_type:=awf/universe  record:=false sensor_model:=sample_sensor_kit vehicle_model:=sample_vehicle scenario:=<path/to/scenario.yaml>  initialize_duration:=5`
6. You can see that pedestrian that is crossing the crosswalk has a predicted path the specified `min_velocity_for_map_based_prediction`

Video: https://drive.google.com/file/d/1nUgun0_PGdurST1mUp4rr-gWAMAZECMP/view?usp=sharing

### Versions

- OS: Ubuntu 22.04
- ROS: Humble
- Autoware: main branch (2022/08/02)

takayuki5168, 2022-09-22:
@satoshi-ota

satoshi-ota, 2022-09-25:
@mitsudome-r @takayuki5168 Sorry I missed this issue. I'll fix it in this week :pray:

closed 2022-10-31 by satoshi-ota