autowarefoundation/autoware_universe #1496: Help needed in matching a PCD map created within a simulator with a Lanelet2 data created in real-world

sglee-morai, 2022-08-02:
### Checklist

- [X] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [X] I've searched other issues and no duplicate issues were found.
- [X] I've agreed with the maintainers that I can plan this task.

### Description

A PCD map for ITRI Campus is created using MORAI SIM: Drive (Actually using AWSIM from TierIV as well).

I put the map along with the original lanelet2 data in a certain path. 

When launching the Autoware using the command below, it is shown in the rviz that PCD data doesn't match the lanelet2 data. 

`ros2 launch autoware_launch autoware.launch.xml map_path:={the_path_including_pcd_map_and_lanelet2_data} vehicle_model:=sample_vehicle sensor_model:=sample_sensor_kit`

### Purpose

This task is required to have an E2E simulator (either MORAI SIM: Drive or AWSIM) prepared for bus ODD development & testing.

Related Issues in the Simulation WG
- [Integration of ITRI campus with ITRI bus and Autoware - MORAI SIM](https://github.com/autowarefoundation/autoware-projects/issues/41) 
- [Integration of ITRI campus with ITRI bus and Autoware - AWSIM](https://github.com/autowarefoundation/autoware-projects/issues/40)

### Possible approaches

I'm guessing there must be a way to specify how to convert the PCD data onto the global coordinate system so that it can be matched onto the lanelet2 map. I'm currently looking into  `map_loader` package, but I cannot find one yet.

### Definition of done

When launching the Autoware using the command below, rviz should show PCD data on top of lanele2 data.

`ros2 launch autoware_launch autoware.launch.xml map_path:={the_path_including_pcd_map_and_lanelet2_data} vehicle_model:=sample_vehicle sensor_model:=sample_sensor_kit`

mitsudome-r, 2022-08-02:
@sglee-morai If you would like to move pointcloud you can either use pcl_tools or CloudCompare
* pcl_tools: install `sudo apt install pcl-tools` and run  pcl_transform_point_cloud
* CloudCompare: https://www.danielgm.net/cc/

VRichardJP, 2022-08-05:
> @sglee-morai If you would like to move pointcloud you can either use pcl_tools or CloudCompare
> 
> * pcl_tools: install `sudo apt install pcl-tools` and run  pcl_transform_point_cloud
> * CloudCompare: https://www.danielgm.net/cc/

I second pcl-tools. In the past I had a similar issue: I created a lanelet2 map from JOSM (georeferenced) and had to match it on the PCD (which was not). You can use the following principle:
1. Identify 2 points (or more) that are both on your lanelet and pcd. For each, find its global (latlon/mgrs/utm) and local (pcd) coordinates
2. compute global and local coordinate of the center point
3. compute average required rotation angle around the center point (stddev should be low)
4. then run something like:
```
# change pcd origin to center point
pcl_transform_point_cloud src dst -trans center_p["local"].x,center_p["local"].y,0
# rotate pcd around the center point
pcl_transform_point_cloud dst dst -axisangle 0,0,1,avg_rot
# translate pcd back to match global coordinates
pcl_transform_point_cloud dst dst -trans -center_p["global"].x,-center_p["global].y,0
# eventually you may have to change the format to binary. You can do this with some python:
# pcl.save(pcl.load(input), output, binary=True)
```

Eventually double check that each point you used as referenced have been moved at the expected coordinate.

sglee-morai, 2022-08-09:
@mitsudome-r @VRichardJP Thanks a lot for providing highly detailed guidelines!

# PCD map sample from the simulator & ITRI Campus Lanelet2 Map captured in the real-world campus

Here is a [download link](https://drive.google.com/file/d/1n3mLfaAUMB1pcAmU7rHvduW3ZJRH5faP/view?usp=sharing) for the initial sample PCD map, created using MORAI SIM: Drive, that you might take a look at if you'd like. 

Btw, the real-world lanelet2 data that I'm trying to match the PCD map to is the ITRI campus lanelet2 map uploaded in the CI/CD pipeline. (I believe I'm not able to share it publicly.)

# Matching
I'm currently trying it out and it seems like I'm almost there. (Blue one is from the real-world one, whereas the red one is from MORAI SIM: Drive)

Due to other issues, I wasn't able to follow the way that  @VRichardJP has suggested, and just give a few numbers that look okay for now. Since I can create another PCD map, which looks much more similar to the real-world one than the current PCD map, I would like to create the map once again and then try matching them.

[image]

sglee-morai, 2022-08-16:
# [22-08-16] Update

### Trying to test, but stuck at another issue (lidar driver)
I created some PCD maps and tried to do some testing before closing this issue.

For testing, I'm trying to run MORAI SIM: Drive with Autwoare to see whether localization is working fine, but I'm struggling in starting a lidar driver for Autoware.Universe 

- Velodyne models and Ouster models of the MORAI SIM: Drive send just like real-world sensors (Same UDP package)
- I'm looking for how to run Velodyne driver and Ouster driver correctly for Autoware.Universe (FYI, for Autoware.Auto, the Velodyne driver was working fine)

xmfcx, 2022-08-16:
https://github.com/tier4/velodyne_vls is being used in the Isuzu bus for velodyne driver.

mitsudome-r, 2022-08-16:
> Is it possible for a lanelet2 map for Autoware can have an arbitrary position (lat, lon) as the origin of the local map coordinate system? (NOT the origin of an MGRS system)
Yes. 
You should be able to provide it as UTM origin by giving as a parameter to the loader with something similar to https://github.com/autowarefoundation/autoware.universe/blob/main/map/map_loader/config/lanelet2_map_loader.param.yaml

sglee-morai, 2022-08-18:
@xmfcx @mitsudome-r Thanks a lot for sharing! 

I was following through how Autoware.Universe launches velodyne drivers, and I found this one: [sample_sensor_kit_launch/launch/lidar.launch.xml](https://github.com/autowarefoundation/sample_sensor_kit_launch/blob/main/sample_sensor_kit_launch/launch/lidar.launch.xml)

Is this the lidar configuration file when launching the Autoware using the [autoware_launch/launch/autoware.launch.xml](https://github.com/autowarefoundation/autoware_launch/blob/main/autoware_launch/launch/autoware.launch.xml) and sample_sensor_kit (the default value) is used for sensor_model argument?

closed 2022-08-23 by sglee-morai