autowarefoundation/autoware_universe #1499: StepPolygon of obstacle_stop_planner is  instable when changing the params of avoidance_module

shulanbushangshu, 2022-08-02:
### Checklist

- [X] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [X] I've searched other issues and no duplicate issues were found.
- [X] I'm convinced that this is not my fault but a bug.

### Description

After changing the param("min_nominal_avoidance_speed"、"nominal_lateral_jerk") of avoidance_module in behavior_path_planner,the StepPolygons in obstacle_stop_planner will be instable in some places when car avoids obstacles.
As follows:
By changing the  "min_nominal_avoidance_speed"、"nominal_lateral_jerk", we can reduce horizontal distance in the act of avoiding obstacles.
case1:
'min_nominal_avoidance_speed':7.0(default)---->3.0
[image]
case2:
'min_nominal_avoidance_speed':7.0(default)---->3.0
'nominal_lateral_jerk':0.2(default)----->0.4
[image]

### Expected behavior

the StepPolygons in obstacle_stop_planner should  be stable whether the parameters of avoidance_module are changed or not .

### Actual behavior

The StepPolygons in obstacle_stop_planner will  be instable when the parameters of avoidance_module are changed.
[image]

### Steps to reproduce

1.Change the param("min_nominal_avoidance_speed"、"nominal_lateral_jerk") of avoidance_module.
2.Start autoware planning simulator 
3. Set initial position and gobal position;
4.Set  obstacle(like car or bus)

### Versions

1.OS: Ubuntun 20.04
2.ROS2: Galactic
3.Autoware: Autoware.universe:main

angry-crab, 2022-08-16:
This is related to #966

shulanbushangshu, 2022-08-23:
yes,the"skip_optimization" is false when simulation.
the static simulation is:
[image]

 the dynamic simulation is:
https://user-images.githubusercontent.com/102840938/186082431-81a46967-e743-4929-b4b6-0b22537c808b.mp4

angry-crab, 2022-08-23:
@shulanbushangshu Thank you for the information. Could you uncomment the following function the try again? 

https://github.com/autowarefoundation/autoware.universe/blob/a9d844ef1f4cf23fffcdbc9cd96c7442fdd51c82/planning/obstacle_avoidance_planner/src/node.cpp#L1404

By default, avoidance_planner would interpolate yaw which causes the issue.

shulanbushangshu, 2022-08-23:
@angry-crab
Yes,I have uncommented the "fillYawInTrajectoryPoint"  and rebuilded.
The result is ok.The "fillYawInTrajectoryPoint" is commented  as it said:"Currently this implementation is removed since this calculation "
How could deal with this function "fillYawInTrajectoryPoint"?

mitsudome-r, 2022-08-23:
@TakaHoribe @takayuki5168 Do you have any comments on this? Should we add a parameter to let the user choose whether to run fillYawInTrajectoryPoint() function?

takayuki5168, 2022-08-25:
@shulanbushangshu @mitsudome-r @angry-crab 
It's ok to set a parameter whether to run fillYawInTrajectoryPoint or not.
But as commented, the calculation cost (mainly spline interpolation with dense spline interpolation) is large, so please set false by default.

I didn't look into how heavy the calculation cost is and how to make it faster.

angry-crab, 2022-08-25:
@takayuki5168 @shulanbushangshu 
This PR should fix the issue. 
#1687

angry-crab, 2022-09-05:
This issue can be closed since the PR is merged.

closed 2022-09-05 by angry-crab