autowarefoundation/autoware_universe #1505: Create a new common package for TensorRT

yukke42, 2022-08-03:
### Checklist

- [X] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [X] I've searched other issues and no duplicate issues were found.
- [X] I've agreed with the maintainers that I can plan this task.

### Description

Since some perception packages implement TensorRT libraries independently, new updates or features are implement for each package. I want to create a new package of TensorRT for the easy managements.

### Purpose

for the easy managements

### Possible approaches

create a new common package

### Definition of done

Complete the description of the following 5 nodes

- [ ]  https://github.com/autowarefoundation/autoware.universe/tree/main/perception/lidar_apollo_instance_segmentation
- [ ]  https://github.com/autowarefoundation/autoware.universe/tree/main/perception/lidar_centerpoint
- [ ]  https://github.com/autowarefoundation/autoware.universe/tree/main/perception/traffic_light_ssd_fine_detector
- [ ]  https://github.com/autowarefoundation/autoware.universe/tree/main/perception/tensorrt_yolo
- [ ]  https://github.com/autowarefoundation/autoware.universe/tree/main/perception/traffic_light_classifier

miursh, 2023-09-14:
@yukke42 Are there any tasks left to do for this?

yukke42, 2023-10-10:
I will close this issue, since I don't have time for it.

closed 2023-10-10 by yukke42