autowarefoundation/autoware_universe #1524: bug scenario_test_runner speed limit doesn't work

beyzanurkaya, 2022-08-04:
### Checklist

- [X] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [X] I've searched other issues and no duplicate issues were found.
- [X] I'm convinced that this is not my fault but a bug.

### Description

I created several scenarios with the scenario editor. these scenarios had speed limit iterations. I tried to ensure that the speed of the vehicle is fixed at 10, 20 and 30 km / h. However, I could not achieve this. Even though I ran the same scenario without changing anything, the vehicle sometimes stayed true to the speed limit I set, but in some cases it did not. I tried the scenarios from  [ODD Repository](https://gitlab.com/autowarefoundation/operational-design-domains/-/tree/master/public_road_bus) as I thought I might have created the scenarios that I created incorrectly. but still faced the same problems. 

### Expected behavior

```
   ControllerAction:
     AssignControllerAction:
       Controller:
         name: ''
         Properties:
            Property:
             - name: maxSpeed
                value: $vel
```
The above section belongs to the file of the scenario. ` value: $vel `  represents the maximum speed of the EGO. EGO must stick to the specified speed limit.

ros2 launch scenario_test_runner scenario_test_runner.launch.py sensor_model:=sample_sensor_kit vehicle_model:=sample_vehicle \
scenario:=/home/beyza/projects/operational-design-domains/public_road_bus/$1 \
architecture_type:=awf/universe launch_rviz:=false launch_autoware:=true
}

### Actual behavior

EGO does not stick to the specified speed limit. If there is a speed limit specified on the lanelet map, EGO accomplish it, otherwise it goes up to 72 km/h.

### Steps to reproduce

1. Edit this command according to you and run 
```
ros2 launch scenario_test_runner scenario_test_runner.launch.py sensor_model:=sample_sensor_kit vehicle_model:=sample_vehicle \
scenario:=path/your/scenario \
architecture_type:=awf/universe launch_rviz:=false launch_autoware:=true
```
2. Observe vehicle velocity.

### Versions

- OS: Ubuntu20.04
- autoware.universe
- ROS2

### Additional context

As you can see I run same senario several times but  the speed limit was different every time.

[image]

[image]

beyzanurkaya, 2022-08-11:
setting  speed limit works commit 970f5172929b9bd25e6d34103965a65753ee1245.

beyzanurkaya, 2022-08-15:
This problem is caused by scenario_simulator_v2. I created this issue in scenario_simulator_v2. You can check [here](https://github.com/tier4/scenario_simulator_v2/issues/856)

xmfcx, 2022-08-16:
This issue belongs to https://github.com/tier4/scenario_simulator_v2/issues/856 so, closing it here.

closed 2022-08-16 by xmfcx