autowarefoundation/autoware_universe #1543: vehicle starts move before the steering angle converged

brkay54, 2022-08-08:
### Checklist

- [X] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [X] I've searched other issues and no duplicate issues were found.
- [X] I'm convinced that this is not my fault but a bug.

### Description

Hi, I limited the `steer_rate_lim_dps` to 5.0 dps in lateral controller. When re-initialize the vehicle, vehicle starts move before steering angle converged.
[video](https://user-images.githubusercontent.com/45468306/183460119-8300f1ed-08e9-4791-8d72-5d2c7a7e5d06.mp4)

### Expected behavior

Vehicle should wait until steering angle converged before move.

### Actual behavior

Vehicle starts move before steering angle converged.

### Steps to reproduce

-

takayuki5168, 2022-08-08:
cc @TakaHoribe

kosuke55, 2022-08-17:
@brkay54 
How about enabling the following options?
- [keep_steer_control_until_converged](https://github.com/autowarefoundation/autoware.universe/blob/main/launch/tier4_control_launch/config/trajectory_follower/lateral_controller.param.yaml#L60)
- [enable_keep_stopped_until_steer_convergence](https://github.com/autowarefoundation/autoware.universe/blob/main/launch/tier4_control_launch/config/trajectory_follower/longitudinal_controller.param.yaml#L9)

takayuki5168, 2022-09-22:
@brkay54 How's it going

brkay54, 2022-10-03:
Hi, I tested in planning_simulator with following parameters: 
[keep_steer_control_until_converged](https://github.com/autowarefoundation/autoware.universe/blob/main/launch/tier4_control_launch/config/trajectory_follower/lateral_controller.param.yaml#L60)
    [enable_keep_stopped_until_steer_convergence](https://github.com/autowarefoundation/autoware.universe/blob/main/launch/tier4_control_launch/config/trajectory_follower/longitudinal_controller.param.yaml#L9)

However, I observed the same behavior([video](https://user-images.githubusercontent.com/45468306/183460119-8300f1ed-08e9-4791-8d72-5d2c7a7e5d06.mp4)). I think it does not cause a problem for real vehicle, but it is a bug for planning_simulator. I suggest that we should reinitialize vehicle if 2D pose estimate is used while vehicle driving. I guess in current implementation it just localize the vehicle.

takayuki5168, 2022-10-03:
I got it. I feel it's better for controllers to have a reset function considering this case only in the simulation.

mitsudome-r, 2022-10-11:
@brkay54 to create a PR for the fix, but he has other tasks with higher priorities at the moment.
Please let me know if anyone can't wait for @brkay54's PR.

brkay54, 2022-11-11:
I investigated the bug, this bug is caused by `mpc_follower`.

If the `steer_rate_lim_dps` is very low and when reinitialize the vehicle while turning, mpc can not solve the problem and it sends the previous control command. Therefore, vehicle starts to move like in [video](https://user-images.githubusercontent.com/45468306/183460119-8300f1ed-08e9-4791-8d72-5d2c7a7e5d06.mp4).

I want to propose that if longitudinal_controller is in stopped state, mpc should not take steer_rate_lim_dps as constraint (or we can make it infinite). I created a branch to test this (not for PR). you can checkout my [branch](https://github.com/brkay54/autoware.universe/tree/mpc-unconstraint-steering-rate-lim) to test it. 

What do you think @maxime-clem @takayuki5168 @kosuke55

kosuke55, 2022-11-14:
@brkay54 Thaks very much, I will check it.

mitsudome-r, 2022-11-22:
@kosuke55 Did you have chance to check it?

kosuke55, 2022-11-22:
@brkay54 
Sorry to be late, I checked in your branch. But I don't understand the expected behavior.
What's the difference from [your main branch video](https://user-images.githubusercontent.com/45468306/183460119-8300f1ed-08e9-4791-8d72-5d2c7a7e5d06.mp4)?

this is a video of using your branch. The vehicle start driving before the steer converged?

https://user-images.githubusercontent.com/39142679/203306381-79a86cfa-28ba-4ff4-92db-0b6c8df7edac.mp4

kosuke55, 2022-11-22:
Is the problem that `m_steer_rate_lim` is so small that the target steer angle changes so little that it doesn't become the real target value right away and is judged to converge at the value that is changing and start driving?
If so, it might be better to compare the difference between `m_ctrl_cmd_prev` and `ctrl_cmd` and judge that it is not converging if it is still changing.

kosuke55, 2022-11-22:
(Also, even if the ego position is changed while the pid is in drive, the car will start immediately because it is still in drive.)

brkay54, 2022-12-04:
Sorry for late reply!

> (Also, even if the ego position is changed while the pid is in drive, the car will start immediately because it is still in drive.)

I checked states of longitudinal controller, when reinitialize the vehicle, state change to emergency, and then to stop. So, cause of this problem is not it.

> Is the problem that `m_steer_rate_lim` is so small that the target steer angle changes so little that it doesn't become the real target value right away and is judged to converge at the value that is changing and start driving?

Actually, the problem is that when reinitialize the vehicle, solver can not solve the problem and sends the previous command to vehicle rather than the target value. 
Like before reinitialize previous cmd = 30.0 degree,
After we reinitialize vehicle, previous_cmd still 30.0 degree but target_value is maybe 0.0 degree ideally but because steering_angle_rate_limit so small, mpc can not solve the problem and sends 30.0 degree again and vehicle starts moving (because before re initialization, current steering was near to 30.0 degree.)

To solve this, I set mpc's steering_rate_limit large as possible if the vehicle in stop state (otherwise it will be what we declared), so It send the (expected) target steering without steering_rate_limit constraint.

> If so, it might be better to compare the difference between `m_ctrl_cmd_prev` and `ctrl_cmd` and judge that it is not converging if it is still changing. 

I tried this with some changes and worked, I will create (draft)PR about it tomorrow.

brkay54, 2022-12-05:
@kosuke55 I created draft PR for this, let discuss there.

mehmetdogru, 2023-05-02:
@brkay54 will update soon.

closed 2023-06-02 by brkay54