autowarefoundation/autoware_universe #1545: the implementation of sigmoid()  in lidar_centerpoint package is not right

WfHit, 2022-08-09:
### Checklist

- [X] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [X] I've searched other issues and no duplicate issues were found.
- [X] I'm convinced that this is not my fault but a bug.

### Description

The implementation of sigmoid()  in lidar_centerpoint package is not right. The current implementation will return value greater than 1, and cause a lot of false positives.

### Expected behavior

__device__ inline float sigmoid(float x) { return 1.0f / (1.0f + expf(-x)); }

### Actual behavior

__device__ inline float sigmoid(float x) { return 1.0f / expf(-x); }

### Steps to reproduce

sigmoid() is defined at 50 line in postprocess_kernel.cu

miursh, 2022-08-09:
@yukke42 Could you please confirm this?

yukke42, 2022-08-09:
@WfHit Thank you for your kind feedback. Cloud you make a PR to fix this bug?

WfHit, 2022-08-09:
OK, I will fix it as soon as possible

PR #1555 (2022-08-10): fix(lidar_centerpoint): fix function sigmoid() 

## Description
fix #1545

## Pre-review checklist for the PR author

The PR author **must** check the checkboxes below when creating the PR.

- [x] I've confirmed the [contribution guidelines].
- [x] The PR follows the [pull request guidelines].

## In-review checklist for the PR reviewers

The PR reviewers **must** check the checkboxes below before approval.

- [ ] The PR follows the [pull request guidelines].

## Post-review checklist for the PR author

The PR author **must** check the checkboxes below before merging.

- [ ] There are no open  [... cut]
closed 2022-08-10 by yukke42