autowarefoundation/autoware_universe #1580: Fail build tvm_utility

doganulus, 2022-08-14:
### Checklist

- [X] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [X] I've searched other issues and no duplicate issues were found.
- [X] I'm convinced that this is not my fault but a bug.

### Description

Fresh Autoware install in Docker fails with the following message.

```
--- stderr: tvm_utility                          
CMake Error at CMakeLists.txt:56 (find_package):
  By not providing "Findtvm_vendor.cmake" in CMAKE_MODULE_PATH this project
  has asked CMake to find a package configuration file provided by
  "tvm_vendor", but CMake did not find one.

  Could not find a package configuration file provided by "tvm_vendor" with
  any of the following names:

    tvm_vendorConfig.cmake
    tvm_vendor-config.cmake

  Add the installation prefix of "tvm_vendor" to CMAKE_PREFIX_PATH or set
  "tvm_vendor_DIR" to a directory containing one of the above files.  If
  "tvm_vendor" provides a separate development package or SDK, be sure it has
  been installed.
```

### Expected behavior

The build must complete without errors in Docker.

### Actual behavior

The build for `twm_utulity` fails.

### Steps to reproduce

1. Follow the steps in [Docker installation](https://autowarefoundation.github.io/autoware-documentation/main/installation/autoware/docker-installation/)

### Versions

- OS: Ubuntu 20.4
- ROS: galactic
- Autoware: main

### Possible causes

`twm_utility` appears a new addition to the repo.

### Additional context

- Do not recommend cloning `main` in the documentation if `main` is not testing with Docker installation.
- Consider recommend a previously tested tag
- Consider how to sync builds with Docker environment better

BonoloAWF, 2022-08-16:
@ambroise-arm @raminterpl  have you encountered this issue in your work with TVM?

ambroise-arm, 2022-08-16:
I believe this happens because tvm_utility was merged recently and the new "tvm_vendor" dependency it introduces wasn't yet part of the docker image. Looking at https://github.com/autowarefoundation/autoware/pkgs/container/autoware-universe/versions?filters%5Bversion_type%5D=tagged it seems a new version of the docker images is being rolled out, so it should be fixed when that completes.

doganulus, 2022-08-17:
I confirm that the build completes as expected using the images tagged `20220817`.

closed 2022-08-17 by doganulus