autowarefoundation/autoware_universe #1581: Deal with U-turn and crossing path

takayuki5168, 2022-08-15:
### Checklist

- [X] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [X] I've searched other issues and no duplicate issues were found.
- [X] I've agreed with the maintainers that I can plan this task.

### Description

Currently, Autoware sometimes cannot deal with following U-turns and crossing paths (edge cases), because of nearest index search algorithm.

[image]
[image]

### Purpose

To deal with the upper edge cases, new algorithm for nearest index search should be implemented.

### Possible approaches

Implement the new algorithm for nearest index search and replace the previous ones with the new one.

### Definition of done

Autoware can deal with the upper edge cases.

takayuki5168, 2022-08-15:
common
https://github.com/autowarefoundation/autoware.universe/pull/1396
https://github.com/tier4/autoware_launch/pull/437
https://github.com/autowarefoundation/autoware.universe/pull/1582

behavior path planner

behavior velocity planner

obstacle avoidance planner

obstacle stop planner
https://github.com/autowarefoundation/autoware.universe/pull/1552

motion_velocity_smoother
https://github.com/autowarefoundation/autoware.universe/pull/1584

velocity controller

mpc follower

maxime-clem, 2022-10-20:
@takayuki5168 can we close this issue ?

takayuki5168, 2022-10-20:
@maxime-clem There is some packages which does not deal with this issue. Therefore, I think this issue should be kept open.