autowarefoundation/autoware_universe #1598: Using INS sensor and add a way to calculate gnss_pose_cov orientation

cyn-liu, 2022-08-16:
### Checklist

- [X] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [X] I've searched other issues and no duplicate issues were found.
- [X] I've agreed with the maintainers that I can plan this task.

### Description

Considering that many users use INS (integrated navigation system) instead of separate GNSS, IMU devices, is it necessary to create an ins.launch.xml file in [sample_sensor_kit_launch](https://github.com/autowarefoundation/sample_sensor_kit_launch/tree/main/sample_sensor_kit_launch/launch) to start the integrated navigation driver?

The launch file also facilitates the input of imu data into [gnss_poser](https://github.com/autowarefoundation/autoware.universe/tree/main/sensing/gnss_poser) , and uses imu orientation to calculate the orientation of gnss_pose_cov, can get a more accurate initial position when using GNSS for localization initialization.

When without navpvt topic input , the original gnss_pose_cov orientation is obtained using position difference, the method is computationally slow and inaccurate, so we want provide an option, using imu orientation as the orientation of gnss_pose_cov,  it is a more general and accurate way.

### Purpose

add INS driver launch file.
add a way to calculate the gnss_pose_cov orientation, using imu data. 

### Possible approaches

add INS driver launch file in sample_sensor_kit_launch package.
add a way to calculate the gnss_pose_cov orientation, using imu orientation, modify in gnss_poser package.

### Definition of done

launch-file-name: ins.launch.xml
gnss_poser:
add input topic: input_topic_imu (sub imu data)
add arg: use_imu_receiver (the flag of whether to use IMU to calculate the orientation of the gnss_pose_cov)

YamatoAndo, 2022-08-17:
We can use `GnssInsOrientationStamped.msg`.
https://github.com/autowarefoundation/autoware_msgs/tree/main/autoware_sensing_msgs

related PRs
https://github.com/autowarefoundation/autoware.universe/pull/1230
https://github.com/autowarefoundation/autoware.universe/pull/1419

cyn-liu, 2022-08-17:
> We can use `GnssInsOrientationStamped.msg`. https://github.com/autowarefoundation/autoware_msgs/tree/main/autoware_sensing_msgs
> 
> related PRs #1230 #1419

thanks, LGTM!
this PR solve the problem of use INS orientation  as GNSS orientation.

closed 2022-08-17 by cyn-liu