autowarefoundation/autoware_universe #1643: "Delete All Objects" function does not work

NorahXiong, 2022-08-22:
### Checklist

- [X] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [X] I've searched other issues and no duplicate issues were found.
- [X] I'm convinced that this is not my fault but a bug.

### Description

"Delete All Objects" function does not work after I updated code today. I clicked "Delete All Objects" button and the button disappeared but the dummy objects remained.
[image]

### Expected behavior

"Delete All Objects" function clears all dummy objects.

### Actual behavior

"Delete All Objects" function failed to clear all dummy objects.

### Steps to reproduce

1. Update code.
2. Add dummy objects and use "Delete All Objects" function.

### Versions

- OS: Ubuntu 20.04
- ROS2: Galactic
- Autoware: Autoware.universe:main

taikitanaka3, 2022-08-22:
@NorahXiong 
Could you share which object still exists?
If it is bus type object it will remain for a bit of time because of multi object tracker.

NorahXiong, 2022-08-22:
> @NorahXiong Could you share which object still exists? If it is bus type object it will remain for a bit of time because of multi object tracker.

I tried pedestrian and car and after several seconds both of them remain.
[image]

taikitanaka3, 2022-08-22:
@h-ohta 
Do you have any idea for this?

h-ohta, 2022-08-22:
@taikitanaka3 Hmm, it works fine in my environment.
@NorahXiong Could you tell me the commit ID which you are using now?

NorahXiong, 2022-08-22:
> @taikitanaka3 Hmm, it works fine in my environment. @NorahXiong Could you tell me the commit ID which you are using now?

commit ae56406ed19661ebda78cd52571f53a2a1ce16f4 (autoware.universe)
I also updated other repositories in autoware.repos.

h-ohta, 2022-08-22:
@NorahXiong Thank you for your information.
I use same commit ID https://github.com/autowarefoundation/autoware.universe/commit/ae56406ed19661ebda78cd52571f53a2a1ce16f4, but problem does not happen in my envioronment...

Could you see some errors in the terminal when the problem occurs?
I would appreciate if you record some videos too.

NorahXiong, 2022-08-22:
> @NorahXiong Thank you for your information. I use same commit ID [ae56406](https://github.com/autowarefoundation/autoware.universe/commit/ae56406ed19661ebda78cd52571f53a2a1ce16f4), but problem does not happen in my envioronment...
> 
> Could you see some errors in the terminal when the problem occurs? I would appreciate if you record some videos too.

I recompiled the whole project and the problem still exists. No error message showed in the terminal(system_monitor not launched). Here is the [video](https://drive.google.com/file/d/1VIyeUN9JIt9rS7oxzzbmWVet6AUYueD6/view?usp=sharing).

taikitanaka3, 2022-08-23:
@NorahXiong 
This is how to use delete all object tool
can you try this?
https://user-images.githubusercontent.com/65527974/186045685-e6415aea-943b-4290-9239-d788a262f73c.mp4

NorahXiong, 2022-08-23:
> @NorahXiong This is how to use delete all object tool can you try this? https://user-images.githubusercontent.com/65527974/186045685-e6415aea-943b-4290-9239-d788a262f73c.mp4

It works. It seems I have misunderstood the usage. Sorry for taking your time. I'll close the issue.

closed 2022-08-23 by NorahXiong