autowarefoundation/autoware_universe #1702: Drivable area from behavior_path_planner is not expanded in this case.

takayuki5168, 2022-08-26:
### Checklist

- [X] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [X] I've searched other issues and no duplicate issues were found.
- [X] I'm convinced that this is not my fault but a bug.

### Description

Drivable area is not expanded in this case, which could be expanded some days ago.
[image]

### Expected behavior

Drivable area is expanded in this case.

### Actual behavior

Not expanded.

### Steps to reproduce

Put a parked vehicle during the first left turn in odaiba map shown in the above figure.

taikitanaka3, 2022-08-26:
@takayuki5168 
Do you include this PR? https://github.com/autowarefoundation/autoware.universe/pull/1688

zulfaqar-azmi-t4, 2022-08-26:
The part that handle this situation is [included](https://github.com/zulfaqar-azmi-t4/autoware.universe/blob/0520de87b981ba9cfa8ff72ff97697806c18d37b/planning/behavior_path_planner/src/scene_module/avoidance/avoidance_module.cpp#L1726:~:text=//%202.%20when%20there%20are%20multiple%20turning%20lanes%20whose%20previous%20lanelet%20is%20the%20same%20in), but for some reason it doesn't work as expected.

closed 2022-08-30 by mitsudome-r