autowarefoundation/autoware_universe #1716: Checking only 4 corners in the drivable area

beyzanurkaya, 2022-08-29:
### Checklist

- [X] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [X] I've searched other issues and no duplicate issues were found.
- [X] I've agreed with the maintainers that I can plan this task.

### Description

When checking whether the footprint of EGO is within the drivable area, only the 4 corners of EGO checked. However, it was observed that although these 4 corners were within the drivable area, the body of EGO was outside the drivable area. In this case, it makes autonomous driving unsafe.

[image]

### Purpose

Making sure that the entire EGO is in the drivable area, otherwise ending autonomous driving.

### Possible approaches

Calculating the intersection of the drivable area and EGO with boost.intersection

### Definition of done

- [ ] Calculate the intersection of the drivable area and EGO.
- [ ] Implemented in the `isOutsideDrivableAreaFrom Rectangle Footprint` function.
- [ ] Test

takayuki5168, 2022-08-30:
@beyzanurkaya Right. We have to fix two packages.
- 1. obstacle avoidance planner
    - calculates the path (where the path does not start from the ego pose as shown in your figure)
    - and check if the calculated path is within the road or not.
- 2. lane departure checker
    - subscribes the predicted path from the lateral controller
    - and check if the predicted path (which starts from the ego pose) is within the road or not.

In your figure, the issue comes from 2. The planned path is within the road, but the predicted path, which was not shown in your figure though, is outside the road.

beyzanurkaya, 2022-08-31:
You're right. I updated description and figure.

mitsudome-r, 2022-09-27:
@mehmetdogru @beyzanurkaya Do you have any updates? If you are busy with bus integration, that's okay. I just want to know the status :)

beyzanurkaya, 2022-09-27:
I actually started working on this. I tried to create a boost polyon by finding the contours of the drivable area.  Then I used boost intersection. But the result was not what I expected. Intersection output was empty.

takayuki5168, 2022-09-27:
@beyzanurkaya 
Just in case, let me comment that
Obstacle avoidance planner, which is in motion planner, can read not lanelet but drivable area as an image. Therefore, do you mean you create boost polygon from an image (drivable area)? I'm worrying a bit about the calculation cost to create boost polygon from an image and calculate intersection via boost.

> But the result was not what I expected. Intersection output was empty.

Please be careful about the rule of boost polygon if you don't know. The order of points of polygon (clockwise or counterclockwise) is determined, and if you create polygon with points in the opposite order, the intersection calculation may fail.

mitsudome-r, 2022-10-11:
@beyzanurkaya Do you have any blockers for this issue/PR?

beyzanurkaya, 2022-10-12:
No, there is no blocker. I did the things in my head and pushed but the intersection result is zero for some reason. I compared the data of the footprint and the drivable area and they contain numerically similar points. 

Here's what I did:
- I converted the OccupancyGrid to grid map and then cv_image
- I found the contour in cv_image and assigned it to a boost polygon
- I assigned the footprint to a boost polygon
- I found their intersection with boost intersection.
At this point, the intersection result is zero.

takayuki5168, 2022-10-13:
boost polygon's points has order (clockwise or anti-clockwise), and if the order is the opposite, the intersection calculation will fail.
This implementation may be helpful.
https://github.com/autowarefoundation/autoware.universe/blob/main/common/tier4_autoware_utils/src/geometry/boost_polygon_utils.cpp

mitsudome-r, 2022-10-25:
@takayuki5168 The [PR](https://github.com/autowarefoundation/autoware.universe/pull/1999) seems to be ready for review. Could you review it or assign someone who's capable of reviewing it?

closed 2023-01-23 by beyzanurkaya