autowarefoundation/autoware_universe #1734: obstacle_avoidance_planner use only vehicle_width/2

beyzanurkaya, 2022-08-30:
### Checklist

- [X] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [X] I've searched other issues and no duplicate issues were found.
- [X] I've agreed with the maintainers that I can plan this task.

### Description

When calculating the corners of the vehicle in the `isOutsideDrivableAreaFromRectangleFootprint` function, half the width of the vehicle is used as the y point.. This will only work correctly when `left_overhang` and `right_overhang` are equal. Using this formula on non-symmetrical vehicles can lead to many miscalculations.

### Purpose

Correct calculation of the vehicle's points on the y-axis

### Possible approaches

When finding the vehicle's position on the y-axis; `vehicle_width = track_widht + left_overhang + right_overhang`. 
`y = vehicle_width/2` formula is used. but when calculating its position on the y-axis; `y = right_overhang + (track_width/2)` and `-y = left_overhang + (track_width/2)` formulas should be used.

### Definition of done

- [ ] Change y position calculation
- [ ] Test

mehmetdogru, 2022-09-05:
assign @hasanzfdn

mehmetdogru, 2022-09-05:
assigned to [@hasanzfdn](https://github.com/hasanzfdn)

hasanzfdn, 2022-09-05:
I would like to work on this issue

PR #1831 (2022-11-15): fix(behavior_velocity_planner): vehicle points lateral calculation 

## Description
closes #1734 
In addition to the issue, the ``` util.cpp ```code has also been changed

## Pre-review checklist for the PR author

The PR author **must** check the checkboxes below when creating the PR.

- [x] I've confirmed the [contribution guidelines].
- [x] The PR follows the [pull request guidelines].

## In-review checklist for the PR reviewers

The PR reviewers **must** check the checkboxes below before approval.

- [ ] The PR follows the [pull request guidelines].

## Post-review checklist for the PR au [... cut]
closed 2022-11-15 by xmfcx