autowarefoundation/autoware_universe #1759: Turning radius is too short on curves

beyzanurkaya, 2022-09-01:
### Checklist

- [X] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [X] I've searched other issues and no duplicate issues were found.
- [X] I'm convinced that this is not my fault but a bug.

### Description

Trajectory from motion planner is outside the drivable area as in the figure below

[image]

### Expected behavior

A path outside the drivable area should not have occurred.

### Actual behavior

Motion planner output is outside of drivable area.

### Steps to reproduce

1. Run planning.simulator
2. Give an initial pose
3. Give a goal pose where EGO should turn

maxime-clem, 2022-09-02:
I was able to reproduce the issue using the planning simulator with the latest autoware (https://github.com/autowarefoundation/autoware/commit/44c70d33825617b56d8de5e6fe921000238238bd).
[image]
I will investigate possible solutions.

maxime-clem, 2022-09-06:
The main issue seems to be the road being too narrow for the dimensions of the vehicle, making it impossible to follow the road while keeping the footprint of the vehicle inside the lane.

**The solution I found is to expand the drivable area**, i.e., the space where the `obstacle_avoidance_planner` can do motion planning.
The drivable area is generated by the `behavior_path_planner` which has parameters to enable expanding the drivable area and to set the left and right expansion distance.
https://github.com/autowarefoundation/autoware.universe/blob/a4078bdf374fc617f5be531e50e892aefbbc53e0/launch/tier4_planning_launch/config/scenario_planning/lane_driving/behavior_planning/behavior_path_planner/lane_following/lane_following.param.yaml#L4-L6

Changing these parameters to the following values allow to drive in the narrow curve.
```yaml
      expand_drivable_area: true
      right_bound_offset: 0.0
      left_bound_offset: 1.0
```
These parameters cannot be changed dynamically and have to be changed in the parameter file loaded at launch time.

https://user-images.githubusercontent.com/78338830/188535432-64aafd69-0da7-4077-b274-e157b6cb906e.mp4

In the video, the following extra topics are shown:
- `DrivableArea` (`/planning/scenario_planning/lane_driving/behavior_planning/path`) with color scheme "costmap".
- `DebugMarkers` (`/planning/scenario_planning/lane_driving/motion_planning/obstacle_avoidance_planner/debug/marker`) only for namespace "mpt_footprints".

mehmetdogru, 2022-09-06:
@maxime-clem 

Thank you for the solution you proposed!

There is one problem related to this approach. If there is an obstacle around and `path_planner/avoidance` module is registered then expanded drivable_area wouldn't be published for mpt to optimize the motion plan since drivable_area is expanded only in `path_planner/lane_following` module.

I will create a separate issue about it. I guess we can close this issue.

xmfcx, 2022-09-06:
> There is one problem related to this approach. If there is an obstacle around and path_planner/avoidance module is registered then expanded drivable_area wouldn't be published for mpt to optimize the motion plan since drivable_area is expanded only in path_planner/lane_following module.

@mehmetdogru can you explain again, @maxime-clem doesn't understand about this worry.

maxime-clem, 2022-09-07:
I opened a discussion about dynamically expanding the drivable area: https://github.com/orgs/autowarefoundation/discussions/2851

mehmetdogru, 2022-09-07:
@xmfcx @maxime-clem 

Sorry for not being clear.

I will attach a video here that explains my worry. 

You can see here clearly that when `path_planner/avoidance` module active (it means there are obstracles around/in the lane) drivable_area is not expanded. That is because drivable_area is extended only in `path_planner/lane_following` module except for some lane_changing cases.

As @maxime-clem suggested implementing a dynamic expansion _not depending on the path_planner modules_ would solve this issue. We can disscuss further in the disscussion.

https://user-images.githubusercontent.com/48479081/188827600-4590d028-34fd-4d2f-9db0-58f80b161c5b.mp4

TakaHoribe, 2022-09-08:
@takayuki5168 Could you check this problem? Seems the Drivable Area generation does not have consistency between behavior_path_planner modules.

takayuki5168, 2022-09-08:
@mehmetdogru 
Static expansion of drivable area is only implemented in lane_following for now.
By implementing it in other modules, you can use the static expansion in any module.

https://github.com/autowarefoundation/autoware.universe/blob/487264478af006e7745b55917b97c5f7d0c6de17/planning/behavior_path_planner/src/scene_module/lane_following/lane_following_module.cpp#L140-L157

mitsudome-r, 2022-09-13:
I think we can discuss the original issue in this discussion: https://github.com/autowarefoundation/autoware/discussions/2851
and @maxime-clem will create a new issue regarding inconsistent drivable area parameter. After the new issue is created, I will close this issue.

maxime-clem, 2022-09-13:
I opened the follow-up issue https://github.com/autowarefoundation/autoware.universe/issues/1854

closed 2022-09-13 by maxime-clem