autowarefoundation/autoware_universe #1761: behavior_path_planner generated path outside of drivable area

beyzanurkaya, 2022-09-01:
### Checklist

- [X] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [X] I've searched other issues and no duplicate issues were found.
- [X] I'm convinced that this is not my fault but a bug.

### Description

If you give to EGO a goal pose a bit inclined , the path is still generated without considering of the width and length of EGO. And this generated path is outside the drivable area.

[image]

[image]

### Expected behavior

If the path to generated is outside the drivable area, the node should not create any path.

### Actual behavior

`behavior_path_planner` creates a path even if the path is outside the drivable area.

[image]

### Steps to reproduce

1. Run planning.simulator
2. Give an initial pose
3. Give a goal pose inclined

takayuki5168, 2022-09-02:
@beyzanurkaya Thank you for your report.
That is because of smooth goal connection in behavior_path_planner.
https://autowarefoundation.github.io/autoware.universe/main/planning/behavior_path_planner/#smooth-goal-connection

behavior_path_planner does not thoroughly consider if the path is inside the lane or not. Sometimes the path from the behavior_path_planner is outside the lanes with a tight right/left turn on a bit narrow road.
To make the path/trajectory inside the lanes (what we call drivable area) is the role of motion planner (obstacle_avoidance_planner).

Therefore, IMO, I think no code change is required for the issue. What do you think.

takayuki5168, 2022-09-02:
If you want, I agree with implementing the function to reject the goal if it's near to the lane boundaries and the angle is weird like in your figure in mission_planner.

beyzanurkaya, 2022-09-06:
I think it makes sense to adding a reject.  because as the vehicle tries to reach this point, a strange trajector occurs at the end of the goal position.

beyzanurkaya, 2022-09-06:
[image]

mehmetdogru, 2022-09-06:
@takayuki5168 

Thank you for your comments.

Is there a reason why we are excepting a goal which its orientation is not aligned with the road lanelet center line whether it is close to the lane boundaries or not?

My proposal is that any goal which is not aligned with the related lanelet center line should be discarded. What is your opinion about this?

mitsudome-r, 2022-09-06:
@mehmetdogru Currently, it is already rejecting the pose with an angle larger than 45 degrees against the direction of the goal lane. 

I could think of two solutions:
* make this a configurable parameter, or
* modify the mission_planning so that it will overwrite the goal pose with lane angle

 @TakaHoribe Any suggestions?

mehmetdogru, 2022-09-07:
@mitsudome-r 

I am on the side of making `th_angle` a config param.
https://github.com/autowarefoundation/autoware.universe/blob/37e8c99425bffff076d0ec2bf302a8a63a3392a6/planning/mission_planner/src/mission_planner_lanelet2/mission_planner_lanelet2.cpp#L210

I can create a PR about it if everybody is ok with it?
@TakaHoribe @mitsudome-r

takayuki5168, 2022-09-07:
@mehmetdogru 
Only with the angle constraint, we still will get an invalid goal since there is almost no constraint for the lateral position of the goal.
Therefore the path from behavior path planner may be out of the drivable area. Is it ok for you?

What I meant by rejecting function is, in addition to the current angle constraint, kind of checking if the vehicle footprint with the goal pose is inside the lane or not.

mehmetdogru, 2022-09-07:
@takayuki5168 

> Only with the angle constraint, we still will get an invalid goal since there is almost no constraint for the lateral position of the goal.

> What I meant by rejecting function is, in addition to the current angle constraint, kind of checking if the vehicle footprint with the goal pose is inside the lane or not.

Yes we will however having the check if vehicle footprint will be inside the lane or not when it is on the goal pose would prevent [`behavior_path_planner/pull_over`](https://autowarefoundation.github.io/autoware.universe/main/planning/behavior_path_planner/#pull-over) working properly I am afraid. Isn't that so?

> Therefore the path from behavior path planner may be out of the drivable area. Is it ok for you?

So since it is a design related topic about behavior_path_planner we can leave it that way for now.

> Anyway I agree to have a parameter for `th_angle`

I guess I will go ahead and create a PR about it then.

takayuki5168, 2022-09-07:
@mehmetdogru 
> would prevent [behavior_path_planner/pull_over](https://autowarefoundation.github.io/autoware.universe/main/planning/behavior_path_planner/#pull-over) working properly

I think there is no problem. There is a road shoulder lane in lanelet map for pull over. That means the vehicle footprint with the goal pose for pull over is inside the lane.
Is this what you are worrying about?

mehmetdogru, 2022-09-07:
@takayuki5168 

Yes that was my worry, thank you :) Then there is no problem checking also the vehicle footprint for goal pose. Then there will be 2 PRs related to this topic for `mission_planner`:

1) make `th_angle` a conf param
2) check vehicle footprint for goal pose and reject if it is outside of the lane boundaries

Do you agree with this?

mitsudome-r, 2022-09-13:
@takayuki5168 Do you have any concerns with @mehmetdogru 's idea? At least, they look reasonable to me.

xmfcx, 2022-09-20:
@takayuki5168 friendly ping

mitsudome-r, 2022-09-27:
@mehmetdogru Are you planning to create a PR for the fix, or would you like someone else to do it?

mehmetdogru, 2022-10-11:
@mitsudome-r 

Sorry for the late reply. We were very busy with the field tests for some time, so couldn't work on some of the issues. 

Yes, I can create a PR about this issue this week.

ismetatabay, 2022-10-18:
@mehmetdogru I created the PR that checks the footprint of the goal and added the th_angle parameter. I'm a bit confused for the parameter part. I defined a parameter struct in the header, considering the future case of adding parameters to the ```mission_planner```. If this is not needed, I can remove that struct and declare it as a variable single parameter.

mehmetdogru, 2022-10-18:
> I created the PR that checks the footprint of the goal and added the th_angle parameter. I'm a bit confused for the parameter part. I defined a parameter struct in the header, considering the future [...]

@ismetatabay

Thanks for working on this.

I would keep it local and declare the struct in `mission_planner_lanelet2.hpp`. You can rename the struct accordingly.

takayuki5168, 2022-10-18:
FYI: There will be a large change on mission_planner, so it may be better to create a PR based on this branch.
https://github.com/autowarefoundation/autoware.universe/pull/1495

ismetatabay, 2022-10-18:
@takayuki5168 @mehmetdogru  Hi, Should I wait for #1495 to merge to avoid conflicts?

xmfcx, 2022-11-01:
This is being worked on: https://github.com/autowarefoundation/autoware.universe/pull/2088

BonoloAWF, 2022-11-03:
> 

@xmfcx  should this issue be closed then?

BonoloAWF, 2022-11-03:
> This is being worked on: #2088

@xmfcx should this issue be closed then?

mehmetdogru, 2022-11-03:
> > This is being worked on: #2088
> 
> @xmfcx should this issue be closed then?

@BonoloAWF it will be closed automatically once #2088 is merged since they are linked.

BonoloAWF, 2022-11-03:
Thanks @mehmetdogru. Moved to next milestone since the PR looks to be ongoing.

PR #2088 (2022-12-13): feat(mission_planner): check goal footprint

## Description

I added checking function for if the goal's footprint is within the lane boundaries. Thus, if the footprint of the given goal extends beyond the lane boundaries, the goal will be rejected. Moreover, while the goal is published, the parameter is added for the maximum angle that the goal makes with the lane to be accepted. 

## Related links
closes #1761 
## Tests performed
Many goals were published on Rviz, it was seen that the goal was rejected when the goal's footprint was outside the lane boundaries. Goal footprint marker publisher [... cut]
closed 2022-12-13 by xmfcx