autowarefoundation/autoware_universe #1784: Timeout happens at getTransform() in SurroundObstacleCheckerNode, and DynamicObject is not used to find surroud objects

takeshi-iwanari, 2022-09-05:
### Checklist

- [X] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [X] I've searched other issues and no duplicate issues were found.
- [X] I'm convinced that this is not my fault but a bug.

### Description

- Where the issue happens: [SurroundObstacleCheckerNode::getNearestObstacleByDynamicObject()](https://github.com/autowarefoundation/autoware.universe/blob/3cae6eca45b36b75c696780e67aabaee05330629/planning/surround_obstacle_checker/src/node.cpp#L367)
- Timeout happens when calling `getTransform()` to get a transform ("map" -> "base_link")
- `getNearestObstacleByDynamicObject()` returns NULL and `nearest_object` becomes NULL. As a result, only `nearest_pointcloud` is used to judge `isStopRequired()` .
- Once the issue happens, timeout always occurs at every call

### Expected behavior

- Timeout should not happen at `getTransform()`, and both `nearest_object` and `nearest_pointcloud` should be valid

### Actual behavior

- Timeout happens 10 / 10, when playing rosbag with `-r 1.0`
- Timeout happens 1 / 10, when playing rosbag with `-r 0.2`
- (It looks this is a timing-dependent problem)

### Steps to reproduce

- Follow [the Rosbag replay simulation tutorial](https://autowarefoundation.github.io/autoware-documentation/main/tutorials/ad-hoc-simulation/rosbag-replay-simulation/)
  - Do not add `-r 0.2` when playing rosbag
  - Add log, so that we can check if timeout happens
    - where to modify: https://github.com/autowarefoundation/autoware.universe/blob/3cae6eca45b36b75c696780e67aabaee05330629/planning/surround_obstacle_checker/src/node.cpp#L376

    ```
      if (!transform_stamped) {
        RCLCPP_WARN(this->get_logger(), "Timeout getNearestObstacleByDynamicObject");   // Added
        return {};
      }
    ```
- You'll see "`Timeout getNearestObstacleByDynamicObject`" log on a terminal once every 500msec

### Versions

- OS: Ubuntu 20.04 (5.15.0-46-generic)
- ROS2: Galactic
- Autoware: b1e2f6ef5982ccbe9434bff49397b2783713cb98
- Autoware.Universe: 3cae6eca45b36b75c696780e67aabaee05330629

### Possible causes

- In short, timeout setting (= 500msec) is too long. While waiting at `getTransform()` on a timer callback, an executor cannot run a subscription callback and the node cannot receive a new topic.
- Callbacks in SurroundObstacleChecker
  - SurroundObstacleChecker has the following callbacks (MultiThreadedExecutor. All callbacks belong to the same callback group)
    - [onDynamicObjects Calback](https://github.com/autowarefoundation/autoware.universe/blob/3cae6eca45b36b75c696780e67aabaee05330629/planning/surround_obstacle_checker/src/node.cpp#L295)
      - Callback to subscribe `/objects` topic
    - [100msec Timer Callback](https://github.com/autowarefoundation/autoware.universe/blob/3cae6eca45b36b75c696780e67aabaee05330629/planning/surround_obstacle_checker/src/node.cpp#L196)
      - Call `getNearestObstacle()` and judge if `isStopRequired()`
      - In `getNearestObstacle()` , `getTransform()` is called using subscribed `/objects` information
      - Timeout setting = 500msec at `getTransform()`
    - (Other callbacks are omitted in this explanation)

- How the issue happens
  1. It looks, `getTransform()` is called with the future time (around 1 sec later) af the first call. As a result, it waits for 500msec
      - I confirmed this by comparing `object_ptr_->header.stamp` and "`ros2 run tf2_ros tf2_echo map base_link`"
  2. While waiting 500msec, new `/objects` topics with latest time are published. However, `onDynamicObjects()` cannot wake-up
      - I confirmed this by adding log at the top of `onDynamicObjects()` and it's not called
      - It's probably because the 100msec timer callback is runnig (waiting for 500msec) (*1)
  3. After 500msec, both the next timer callback and `onDynamicObjects()` callback are ready. However, A timer callback has higher priority than a subscription callback, so the 100msec timer callback will wake-up again and `onDynamicObjects()` callback keeps not running
      - Reference: https://docs.ros.org/en/galactic/Concepts/About-Executors.html#scheduling-semantics
  4. `getTransform()` is called with older timestamp contained in older `/objects` msg, but the corresponding transform is not found and it will wait for 500msec again  (*2)
  5. Go to 2.

- Possible Solutions
  - Assign different callback group for onDynamicObjects Calback and 100msec Timer Callback
    - It may cause another bug in the future, because these functions share variables
  - Decrease timeout setting from 500msec to 50msec so that `onDynamicObjects()` can wake-up and update `/objects` msg
    - I confirmed this solved the issue
    - It can be any as long as less than 100msec
    - 50msec works to solve this problem, but may not be appropriate for another situatiom

- Unclear stuffs
  - *1:  `getTransform()` calls `lookupTransform()` in `buffer.cpp` in ROS2 code. In the code, `sleep_for(10msec)` is called. So, thread switching should happen, but it looks the timer callback occupies the executor
  - *2: `tf_buffer_` should be able to contain /tf for 10 seconds. So, a transform with older timestamp should be found. It looks `transform_listener` doesn't wake up and cannot update buffer (not yet confirmed)

maxime-clem, 2022-10-20:
@piotr-zyskowski-rai do you have any update on this issue ?

piotr-zyskowski-rai, 2022-12-05:
@maxime-clem Sorry for the very late response. @lchojnack  will start to work on this issues after he finishes current tasks

takeshi-iwanari, 2023-03-15:
@lchojnack 
Thank you for following up.
I confirmed the issue doesn't happen with [the latest version (2023/03/15)](https://github.com/autowarefoundation/autoware.universe/tree/b93dbc5c3dd5bd497555c5a12a9b2cdd679b5b56).

Details:

> i. It looks, 'getTransform()' is called with the future time (around 1 sec later) af the first call. As a result, it waits for 500msec 

In the past [`object_ptr_->header.stamp`](https://github.com/autowarefoundation/autoware.universe/blob/b93dbc5c3dd5bd497555c5a12a9b2cdd679b5b56/planning/surround_obstacle_checker/src/node.cpp#L370) was the future time compared to transform map time  and it caused the issue. `object_ptr_->header.stamp` is valid in the latest version and timeout doesn't occur anymore.

closed 2023-03-15 by takeshi-iwanari