autowarefoundation/autoware_universe #1793: The path of behavior_path_planner is out of lane when obstacle is near intersection

shulanbushangshu, 2022-09-06:
### Checklist

- [X] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [X] I've searched other issues and no duplicate issues were found.
- [X] I'm convinced that this is not my fault but a bug.

### Description

When the obstacle is near the intersection,the path of behavior_path_planner is out of lane.
The simulation is shown as below:
[image]

### Expected behavior

The path of behavior_path_planner should not be out of lane.The path should have no avoiding behavior in intersection.

### Actual behavior

The path of behavior_path_planner is out of lane when obstacle is near intersection

### Steps to reproduce

1.Start autoware planning simulator
2. Set initial position and gobal position;
3.Set obstacle(like car or bus)

### Versions

1.OS: Ubuntun 20.04
2.ROS2: Galactic
3.Autoware: Autoware.universe:main

### Possible causes

There is no check function in avoid_module of behavior_path_planner about intersection

zulfaqar-azmi-t4, 2022-09-06:
Hi @shulanbushangshu,

Is it possible to turn on both drivable area marker and avoidance marker and share it with us?

shulanbushangshu, 2022-09-07:
@zulfaqar-azmi-t4 
Near the intersection,the lanes may be separated.The drivable area in simulation is shown as below:
[image]

Maybe there should be function to deal with  obstacle near intersection.

mehmetdogru, 2022-09-07:
@shulanbushangshu 

As a workaround -not just for intersection but for all the lanes-, if you would want ego vehicle to stay its lane while avoiding an obsacle you can switch following parameters to `false`:

https://github.com/autowarefoundation/autoware.universe/blob/c211845a40947212fee20b357d0bb49e2c6f77b0/launch/tier4_planning_launch/config/scenario_planning/lane_driving/behavior_planning/behavior_path_planner/avoidance/avoidance.param.yaml#L10-L11

shulanbushangshu, 2022-09-07:
@mehmetdogru 
Yes,the drivable area can be reduced or expanded.However, you cannot change the flags automatically when you run code.And in the test map,the separated lanes  is "same_direction".And I think the obstacle near the intersection should be dealed with.

zulfaqar-azmi-t4, 2022-09-07:
@shulanbushangshu 
Although in the avoidance module side, the trajectory is out of the lane, the `obstacle_avoidance_planner` will create the suitable trajectory by constraining the path within the drivable area. Example is as shown in below videos

https://user-images.githubusercontent.com/93502286/188820702-9583623b-28f0-4fdd-b2ef-e6c02c3d1060.mp4

https://user-images.githubusercontent.com/93502286/188820716-7704eb15-1a91-4365-9fef-39b459144b02.mp4

shulanbushangshu, 2022-09-07:
@zulfaqar-azmi-t4 
Yes, I see what you mean.Maybe it is not a problem.
But at the intersection, if the stopping point is not good, the behavior path will block other vehicles of other lanes.
And in some special scenes like intersection,whether it must avoid the obstacles(Maybe the car can just stop).

xmfcx, 2022-09-20:
Maybe we can add an option like:
- disable_swerving_before_intersection
- distance_from_intersection_to_disable_swerving (rename pls)

So the planner would look ahead of x meters to check if a lanelet with `turn_direction` tag and disable the obstacle avoidance for the vehicle.

satoshi-ota, 2022-09-22:
Hi, @xmfcx Thank you for your ideas ! And it looks good to me :smile: 

On the other hand, we (TIER4) would like to improve avoidance module and fix these improper behaviors by following policies:

- avoidance module should not output unfollowable shift path like above figures
- update logic for shift path generation and make shift path inside drivable area as far as possible (in progress)
- check shift length of whole shift path points and guard improper path

xmfcx, 2022-10-04:
@mehmetdogru for @satoshi-ota 's 2nd point

> update logic for shift path generation and make shift path inside drivable area as far as possible (in progress)

Can you create a PR for limiting the shift-points within lane shoulders?

shulanbushangshu, 2022-10-08:
1.Add check function
Maybe we can detect whether the offset points on the path exceed the lanes from two aspects:
(1) When generating the initial offset points, we can generate the local path according to the local offset point, and then judge whether the points on the locally generated path exceeds the lanes. If it exceeds the lane, we can directly remove the obstacle (it will not avoid the obstacle). (The logic is relatively simple, but local  paths need to be calculated).
(2) After the module finally generates  the  whole path, judge whether the points with offset exceeds the lane. If path exceeds the lanes, the offset points of the path are recalculated by information of the current offset points  to obtain reasonable offset points. (The logic of recalculating the offset points may be complex)
2. In some special scenes 
In some special scenarios, we can add indicators, such as distance, to ignore obstacles.

closed 2022-11-16 by shulanbushangshu