autowarefoundation/autoware_universe #1812: The paths of behavior module and avoidance module are separated when the "enable_avoidance_over_opposite_direction" is false

shulanbushangshu, 2022-09-08:
### Checklist

- [X] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [X] I've searched other issues and no duplicate issues were found.
- [X] I'm convinced that this is not my fault but a bug.

### Description

The paths of behavior module and avoidance module are separated when the "enable_avoidance_over_opposite_direction" is false.The drivable area can't use the "same_direction" lanes when "enable_avoidance_over_same_direction" is true. The drivable area is restricted to the current Lanes.
The simulation is shown as below:

[image]

### Expected behavior

1.When "enable_avoidance_over_opposite_direction" is false, if the drivable area can't use the "same_direction" lanes by design, the avoidance behavior should not happen. 
2.When "enable_avoidance_over_opposite_direction" is false,if the drivable area can use the "same_direction" lanes by design, the drivable area should be changed.
The paths of behavior module and avoidance module should not be separated .

### Actual behavior

When "enable_avoidance_over_opposite_direction" is false,  the drivable area can't use the "same_direction" lanes.
The paths of behavior module and avoidance module are separated .

### Steps to reproduce

1.Set "enable_avoidance_over_opposite_direction" to false
2.Start autoware planning simulator
3. Set initial position and gobal position;
4.Set obstacle(like car or bus)

### Versions

1.OS: Ubuntun 20.04
2.ROS2: Galactic
3.Autoware: Autoware.universe:main

zulfaqar-azmi-t4, 2022-09-08:
Sorry for the mistake, I have made to PR to solve this issue.

shulanbushangshu, 2022-09-09:
@zulfaqar-azmi-t4
Yes,the bug is fixed and I will close the issue.

closed 2022-09-09 by shulanbushangshu