autowarefoundation/autoware_universe #1850: endless recursion issue in RouteHandler

VRichardJP, 2022-09-13:
### Checklist

- [X] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [X] I've searched other issues and no duplicate issues were found.
- [X] I'm convinced that this is not my fault but a bug.

### Description

I am using the following lanelet route:

The vehicle starts from the blue area, makes a clockwise loop and comes back to the blue area in the opposite direction. Half-way in the loop, I see the planning container crashes because RouteHandler gets stuck in an endless recursion [getRightMostLinestring](https://github.com/autowarefoundation/autoware.universe/blob/e28c1698420c4b13b836f1a04f96591342e24685/planning/route_handler/src/route_handler.cpp#L961) -> [getLeftMostLinestring](https://github.com/autowarefoundation/autoware.universe/blob/e28c1698420c4b13b836f1a04f96591342e24685/planning/route_handler/src/route_handler.cpp#L988) -> [getRightMostLinestring](https://github.com/autowarefoundation/autoware.universe/blob/e28c1698420c4b13b836f1a04f96591342e24685/planning/route_handler/src/route_handler.cpp#L961) -> ...

The particularity of my lanelet2 map is that there are 2 lanelets at the blue area, sharing the same linestrings and going opposite directions (one for the start and one for the end). So if you think about it, the both lanelets are the left and right neighbor of each other... so if the code is not careful, it is easy to get stuck in a loop, searching for the most left/right something of one of the 2 lanelets.

I know one way to fix that is not to share the linestring (or use 'one_way'=false attribute?). But couldn't we make the code more robust?

### Expected behavior

planner does not crash

### Actual behavior

endless recursion

### Steps to reproduce

1. have 2 opposite lanelets sharing same linestring

zulfaqar-azmi-t4, 2022-09-13:
@VRichardJP
The way the function is written is intended for the user to properly follow the lanelet2 format, so it that any "bug" related to library can be traceback to the library. Therefore, I do recommend to as the false attribute on the one way tag.
 
However, it is also true that the code should be more robust. So we can think of some way to address this issue.

Can you share the lanelet and pcd to us so that we can take a look a look and use them for testing?

VRichardJP, 2022-09-13:
Here is the lanelet map I used (changed to .txt to upload on github):
[lanelet2_map.txt](https://github.com/autowarefoundation/autoware.universe/files/9554161/lanelet2_map.txt)

NorahXiong, 2022-09-20:
I've seen a similar discussion [here](https://github.com/orgs/autowarefoundation/discussions/2600#discussion-4173579).
Maybe you should set the "one_way" tag as "no".
You can try this [file](https://github.com/autowarefoundation/autoware.universe/files/9604750/one_way.zip).

mitsudome-r, 2022-09-20:
I believe current planner doesn't support bi-directional lanes. However, as a work around, you can create lanes in opposite direction that does **not** share linestrings.

zulfaqar-azmi-t4, 2022-09-22:
@VRichardJP 
Could you check the recursion issue status.
This PR (https://github.com/autowarefoundation/autoware.universe/pull/1901) should be able  to solve them.

xmfcx, 2022-10-04:
@VRichardJP can you check if https://github.com/autowarefoundation/autoware.universe/pull/1901 solves this issue?