autowarefoundation/autoware_universe #1879: Align TVM packages with ModelZoo changes

ambroise-arm, 2022-09-15:
### Checklist

- [X] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [X] I've searched other issues and no duplicate issues were found.
- [X] I've agreed with the maintainers that I can plan this task.

### Description

The neural network artifacts built for TVM as part of the [ModelZoo](https://github.com/autowarefoundation/modelzoo) CI are changing in layout. The change has been discussed in https://github.com/autowarefoundation/modelzoo/issues/63 and https://github.com/autowarefoundation/autoware/discussions/2508. The Autoware packages need to align with the change.

ModelZoo artifacts changed from being held in a single archive to one archive per model/backend combination. It allows users to only download needed archives.

This task depends on https://github.com/autowarefoundation/autoware.universe/issues/628 being completed first.

The prebuilt asset rework will be needed for the neural networks added in ModelZoo for https://github.com/autowarefoundation/autoware.universe/issues/908 to be available in Autoware.

### Purpose

Reduce bandwidth usage.

Have a better versioning of the models.

### Possible approaches

Change from all models being handled by the "neural_networks_provider" package to models being downloaded by packages that need them.

Leverage the newly added versioning of the prebuilt models.

### Definition of done

- [ ] Rework prebuilt assets management
- [ ] Model descriptors versioning

angry-crab, 2022-09-28:
Please correct me if I'm wrong. It seems that the currect ci compiles the tvm library and uploads it. The library will be downloaded when building the package. But these packages only contain inference functionality. To make full use of `tvm` utility, I think it is necessary to create pre and postprocessing operations with tvm. For example, this [page](https://tvm.apache.org/docs/tutorial/tensor_expr_get_started.html) introduces a simple way to create tvm operations which can be optimized further using different `schedule` modules. Therefore, in the near future, I'm think of moving the whole pipeline to tvm implementation as opposed to current implementation of c++. Do you think it is necessary or not. If the answer is yes, how should we add corresponding operations to the package? Please let me know you thoughts.

ambroise-arm, 2022-09-28:
You are correct in the description of the inference functionality.

Note that I have never used those `tvm operations`, so I don't know how difficult, or how appropriate it would be for what we want to achieve. 

From my point of view, using `tvm operations` in the pre/post processing steps is not necessary. But my motivation for using TVM is mainly based on portability of the hardware, not so much performance. So I am happy with C++ for pre/post processing, but it would most likely be more performant to use code that can be compiled for a dedicated HW backend.
Maybe a first step could be benchmarking `lidar_centerpoint` with C++ processing to see if it is acceptable or not.

If we do choose to use `tvm operations`, it may be worth revisiting the way things get compiled. The `tvm operations` would need to be compiled for the same target backend as the inference, so it may then be interesting to move the compilation of the neural network away from the ModelZoo CI and into the build step of the Autoware package.

closed 2022-11-11 by angry-crab