autowarefoundation/autoware_universe #1904: accel_brake_map_calibrator sometimes produce invalid map file

VRichardJP, 2022-09-17:
### Checklist

- [X] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [X] I've searched other issues and no duplicate issues were found.
- [X] I'm convinced that this is not my fault but a bug.

### Description

I went through the accel/brake map calibration of one vehicle today. After driving with a joystick for a while I saved the file with the `/accel_brake_map_calibrator/update_map_dir` service.

But upon testing, the `raw_vehicle_cmd_converter` would always crash on loading the generated data (in `interpolation_utils::validateKeys`). I inspected the file and saw that sometimes the column values were the same over consecutive rows. Like this:
```
0.3,-1.48,-1.5,-1.48,...
0.4,-1.48,-1.5,-1.85,...
```
I manually edited the value so that accel values are strictly increasing. Like so:
```
0.3,-1.48,-1.5,-1.48,...
0.4,-1.49,-1.51,-1.85,...
```
And the `raw_vehicle_cmd_converter` is not crashing anymore.

I don't know whether it is normal for the interpolation to crash when values are not strictly increasing. But either the interpolation of the calibrator should be fixed

### Expected behavior

accel/brake maps generated by the `accel_brake_map_calibrator` should always be valid.

### Actual behavior

sometimes incompatible maps are generated

### Steps to reproduce

1. calibrate

### Versions

- Ubuntu 20.04
- ROS2 galactic
- master (~1 month old)

takayuki5168, 2022-09-22:
@VRichardJP Thanks for your bug report! I guess this should be fixed. In this case, the interpolated function has a discontinuity.

@TakaHoribe How about asking this bug fix to a PT student. Or just setting `help_wanted` sounds fine.

VRichardJP, 2022-09-26:
example of brake_map.csv generated from the calibrator:
[brake_map.csv](https://github.com/autowarefoundation/autoware.universe/files/9646320/brake_map.csv)

0.6, 0.7 and 0.8 rows have not been calibrated (too strong brake), most values are the same and `raw_vehicle_cmd_converter` crashes at runtime when using this map.

xmfcx, 2022-12-13:
@TakaHoribe @takayuki5168 any updates on this issue?