autowarefoundation/autoware_universe #1907: Can not set initial pose in planning simulator after code update

NorahXiong, 2022-09-19:
### Checklist

- [X] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [X] I've searched other issues and no duplicate issues were found.
- [X] I'm convinced that this is not my fault but a bug.

### Description

I failed to set initial pose in planning simulator after updating the code(5f7fbd74171ef90b5f84c35e9e0088bb8f71690e).

https://user-images.githubusercontent.com/103234047/190976377-fe9b7f81-f781-4391-965e-e7a3931f2cd0.mp4

### Expected behavior

Planning simulator works normally.

### Actual behavior

Can not set initial pose in planning simulator.

### Steps to reproduce

Update the project and try to set initial pose in planning simulator.

### Versions

- OS: Ubuntu 20.04
- ROS2: Galactic
- Autoware: Autoware.universe:main

### Possible causes

In PR #1500, topic "/initialpose" is changed to "/initialpose3d" but no message is sent on topic "/initialpose3d" when simulating. Maybe there are some relating parameters need modification at the same time.

### Additional context

I changed the rviz_default_plugins/SetInitialPose [topic](https://github.com/autowarefoundation/autoware_launch/blob/main/autoware_launch/rviz/autoware.rviz#L1710) from "/initialpose" to "/initialpose3d" and the simulator worked. Anyway, it seems not a proper fix as the pose set through rviz_default_plugins still missing z axis value(z value is always 0).

VRichardJP, 2022-09-19:
I'm facing the same problem after updating autoware. I think autoware's launch files should run this module https://github.com/autowarefoundation/autoware.universe/tree/main/system/default_ad_api_helpers/ad_api_adaptors

isamu-takagi, 2022-09-20:
Sorry, it was because these PRs were not merged at the same time. It's fixed now.
- https://github.com/autowarefoundation/autoware.universe/pull/1900
- https://github.com/autowarefoundation/autoware_launch/pull/88

NorahXiong, 2022-09-20:
> Sorry, it was because these PRs were not merged at the same time. It's fixed now.
> 
> * [feat: move adapi rviz adaptor  #1900](https://github.com/autowarefoundation/autoware.universe/pull/1900)
> * [feat: move adapi rviz adaptor  autoware_launch#88](https://github.com/autowarefoundation/autoware_launch/pull/88)

Planning simulator works well now. Thank you.