autowarefoundation/autoware_universe #1912: Planning simulator doesn't seem to work on Arm

kasperornmeck, 2022-09-20:
### Checklist

- [X] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [X] I've searched other issues and no duplicate issues were found.
- [X] I'm convinced that this is not my fault but a bug.

### Description

When following the instructions from
* https://autowarefoundation.github.io/autoware-documentation/main/installation/autoware/docker-installation/
* https://autowarefoundation.github.io/autoware-documentation/main/tutorials/scenario-simulation/planning-simulation/installation/
to run the planning simulation, the container doesn't start.

### Expected behavior

For the container to launch using `rocker`.

### Actual behavior

Container doesn't launch.

### Steps to reproduce

### Clone repo and pull container
```
ubuntu@ava:~$ git clone https://github.com/autowarefoundation/autoware.git
ubuntu@ava:~$ docker pull ghcr.io/autowarefoundation/autoware-universe:galactic-20220915-prebuilt-cuda-arm64
ubuntu@ava:~$ docker images
REPOSITORY                                     TAG                                     IMAGE ID       CREATED         SIZE
ghcr.io/autowarefoundation/autoware-universe   galactic-20220915-prebuilt-cuda-arm64   399c11688e53   5 days ago      18.2GB
ubuntu@ava:~$ rocker -e LIBGL_ALWAYS_SOFTWARE=1 --x11 --user --volume $HOME/autoware -- ghcr.io/autowarefoundation/autoware-universe:galactic-20220915-prebuilt-cuda-arm64
```

### Inside container
```
ubuntu@cdf526db27a5:~$ cd autoware/
ubuntu@cdf526db27a5:~/autoware$ mkdir src
ubuntu@cdf526db27a5:~/autoware$ vcs import src < simulator.repos
ubuntu@cdf526db27a5:~/autoware$ sudo apt update
ubuntu@cdf526db27a5:~/autoware$ rosdep update
ubuntu@cdf526db27a5:~/autoware$ rosdep install -y --from-paths . --ignore-src --rosdistro $ROS_DISTRO
```

### Error message
```
...
executing command [sudo -H apt-get install libembree-dev]
Reading package lists... Done
Building dependency tree       
Reading state information... Done
E: Unable to locate package libembree-dev
ERROR: the following rosdeps failed to install
  apt: command [sudo -H apt-get install libembree-dev] failed
...
```

### Versions

- Ubuntu 20.04 LTS
- ROS2
- ghcr.io/autowarefoundation/autoware-universe:galactic-20220915-prebuilt-cuda-arm64

### Possible causes

Seems as though `libembree-dev` isn't available on Arm.

kasperornmeck, 2022-09-21:
Proposed fix:
1. install `libembree-dev` manually
2. add `--skip-keys=embree` to the `rosdep install` command
3. see if CMake can find the package when running `colcon build`

mitsudome-r, 2022-09-21:
@ito-san Already ran planning simulator on ARM machine.
https://github.com/autowarefoundation/autoware-projects/issues/32#issuecomment-1217781154 

@ito-san Do you have instructions to reproduce your test on Open AD Kit?

mitsudome-r, 2022-09-21:
@kasperornmeck There seems to be an instruction for Open AD Kit installation. https://autowarefoundation.github.io/open-ad-kit-docs/latest/version-2.0/start-guide/
Have you checked this out?

evshary, 2022-09-22:
@ito-san I tried to run Open AD Kit v2.0 based on [this guide](https://github.com/autowarefoundation/open-ad-kit-docs/blob/proposal_v2.0/docs/start-guide/index.md)
In this guide, Autoware is running on AVA platform (Yocto-based) and the scenario_simulator_v2 is running on my x86 host, which is no problem.
Then I also want to try scenario_simulator_v2 on AVA platform, so use your docker container.
But I meet the following error message:

[image]

Not sure whether it's related to the dependency issue, maybe I need to fetch the latest scenario_simulator_v2 code and try again.
Do you have any suggestions about this error?

evshary, 2022-09-23:
@ito-san Thank you for the explanation.
After viewing the output, I found I have the wrong path of the map in the docker container.
It works perfectly after correcting the path.
Thank you!

xmfcx, 2022-10-04:
@kasperornmeck @evshary I am closing this issue since it's fixed, please reopen it if I understood it wrong.

closed 2022-10-04 by xmfcx