autowarefoundation/autoware_universe #1913: Issues launching NVIDIA enabled container on Arm

kasperornmeck, 2022-09-20:
### Checklist

- [X] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [X] I've searched other issues and no duplicate issues were found.
- [X] I'm convinced that this is not my fault but a bug.

### Description

Issue launching CUDA-enabled container on Arm when using`rocker`.

### Expected behavior

For container to launch using `rocker`.

### Actual behavior

Container doesn't start.

### Steps to reproduce

### Launch container using `rocker`
```
ubuntu@ava:~$ rocker --nvidia --x11 --user --volume $HOME/autoware -- ghcr.io/autowarefoundation/autoware-universe:galactic-20220915-prebuilt-cuda-arm64
```
### Error log
```
Extension volume doesn't support default arguments. Please extend it.
Active extensions ['nvidia', 'volume', 'x11', 'user']
Step 1/12 : FROM python:3-slim-stretch as detector
 ---> 839514fd3bcb
Step 2/12 : RUN mkdir -p /tmp/distrovenv
 ---> Using cache
 ---> bc21c25e506c
Step 3/12 : RUN python3 -m venv /tmp/distrovenv
 ---> Using cache
 ---> fea14b2bc265
Step 4/12 : RUN apt-get update && apt-get install -qy patchelf binutils
 ---> Using cache
 ---> 0c32cbbc1445
Step 5/12 : RUN . /tmp/distrovenv/bin/activate && pip install distro pyinstaller==4.0 staticx==0.12.3
 ---> Running in 5e0103ccc121
Collecting distro
  Downloading https://files.pythonhosted.org/packages/e1/54/d08d1ad53788515392bec14d2d6e8c410bffdc127780a9a4aa8e6854d502/distro-1.7.0-py3-none-any.whl
Collecting pyinstaller==4.0
  Downloading https://files.pythonhosted.org/packages/82/96/21ba3619647bac2b34b4996b2dbbea8e74a703767ce24192899d9153c058/pyinstaller-4.0.tar.gz (3.5MB)
  Installing build dependencies: started
  Installing build dependencies: finished with status 'done'
  Getting requirements to build wheel: started
  Getting requirements to build wheel: finished with status 'done'
    Preparing wheel metadata: started
    Preparing wheel metadata: finished with status 'done'
Collecting staticx==0.12.3
  Downloading https://files.pythonhosted.org/packages/92/ff/d9960ea1f9db48d6044a24ee0f3d78d07bcaddf96eb0c0e8806f941fb7d3/staticx-0.12.3.tar.gz (68kB)
    Complete output from command python setup.py egg_info:
    Traceback (most recent call last):
      File "<string>", line 1, in <module>
      File "/tmp/pip-install-kk2il96y/staticx/setup.py", line 4, in <module>
        from wheel.bdist_wheel import bdist_wheel
    ModuleNotFoundError: No module named 'wheel'
    
    ----------------------------------------
Command "python setup.py egg_info" failed with error code 1 in /tmp/pip-install-kk2il96y/staticx/

You are using pip version 19.0.3, however version 22.2.2 is available.
You should consider upgrading via the 'pip install --upgrade pip' command.

Removing intermediate container 5e0103ccc121
no more output and success not detected
Failed to build detector image
WARNING unable to detect os for base image 'ghcr.io/autowarefoundation/autoware-universe:galactic-20220915-prebuilt-cuda-arm64', maybe the base image does not exist
```

### Versions

- Ubuntu 20.04 LTS
- ROS2
- ghcr.io/autowarefoundation/autoware-universe:galactic-20220915-prebuilt-cuda-arm64

### Possible causes

Container seems to be prevented from starting by:
```
ModuleNotFoundError: No module named 'wheel'
```

### Additional context

When replacing `--nvidia` with `-e LIBGL_ALWAYS_SOFTWARE=1` the container starts.
```
ubuntu@ava:~$ rocker -e LIBGL_ALWAYS_SOFTWARE=1 --x11 --user --volume $HOME/autoware -- ghcr.io/autowarefoundation/autoware-universe:galactic-20220915-prebuilt-cuda-arm64
```

mitsudome-r, 2022-10-11:
Let's discuss this in Open AD Kit WG.

mitsudome-r, 2022-10-11:
cc. @ito-san

xmfcx, 2022-12-13:
Is this issue still ongoing?

kasperornmeck, 2022-12-19:
@xmfcx 

This will be a task which Open AD Kit will tackle in a different way, closing this issue.

closed 2022-12-19 by kasperornmeck