autowarefoundation/autoware_universe #1929: ARM64 image don't work in Jetson and Nvidia Orin

leonlee723, 2022-09-21:
### Checklist

- [X] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [X] I've searched other issues and no duplicate issues were found.
- [X] I'm convinced that this is not my fault but a bug.

### Description

docker run -it --gpus all --privileged --net=host --runtime=nvidia -e DISPLAY=$DISPLAY ghcr.io/autowarefoundation/autoware-universe:galactic-20220915-cuda-arm64

docker: Error response from daemon: failed to create shim: OCI runtime create failed: nvidia-container-runtime did not terminate successfully: exit status 1: time="2022-09-21T19:25:45+08:00" level=error msg="failed to create NVIDIA Container Runtime: failed to construct OCI spec modifier: requirements not met: unsatisfied condition: cuda>=11.6 (cuda=11.4)"

### Expected behavior

ARM64 image can not work in Jetson.
As your know, Jetson platform must install by SDK Manager of Nvidia, bug SDK Manager just install cuda 11.4. 

### Actual behavior

docker: Error response from daemon: failed to create shim: OCI runtime create failed: nvidia-container-runtime did not terminate successfully: exit status 1: time="2022-09-21T19:25:45+08:00" level=error msg="failed to create NVIDIA Container Runtime: failed to construct OCI spec modifier: requirements not met: unsatisfied condition: cuda>=11.6 (cuda=11.4)"

### Steps to reproduce

1.docker run -it --gpus all --privileged --net=host --runtime=nvidia -e DISPLAY=$DISPLAY ghcr.dockerproxy.com/autowarefoundation/autoware-universe:galactic-20220915-cuda-arm64

### Versions

- OS: Linux ubuntu 20.4
- Docker image tag: ghcr.io/autowarefoundation/autoware-universe:galactic-20220915-cuda-arm64

mitsudome-r, 2022-09-27:
Did you take a look on past discussion posts about installing Autoware on Jetson platform?
* https://github.com/autowarefoundation/autoware/discussions/2744
* https://github.com/autowarefoundation/autoware/discussions/2490
* https://github.com/autowarefoundation/autoware/discussions/2601
* https://github.com/autowarefoundation/autoware/discussions/2573

xmfcx, 2022-10-04:
@NilaySener what was your experience like getting Autoware run on AGX Orin? Did you manage to have it working?

NilaySener, 2022-10-10:
I was able to run it without any problems with the images and steps in [this test report.](https://github.com/autowarefoundation/autoware-projects/files/9420690/Test.report.-.Run.AW.universe.with.OpenADKit.v1.0.on.Orin.pdf)

leonlee723, 2022-10-12:
@NilaySener thanks your feedback. I read your pdf. I find you use galactic-latest-prebuilt-arm64, not galactic-latest-cuda-arm64. I don't galactic-latest-prebuilt-arm64 whether support cuda in orin for example PCL. I'll try it. Close with comment.

oguzkaganozt, 2023-09-26:
Hi @greensky0802, could you share how did you manage to compile autoware.universe inside Ubuntu 20.04 ? Because CUDA version is 11.4 in jetson packages which is not sufficient as the current CUDA version is 11.6 in universe. Also rosdep installations may fail due to Ubuntu version differences (20.04 and 22.04).

greensky0802, 2023-10-09:
Hi @oguzkaganozt Sorry for the delay. The first step is to fully install JetPack: 5.1.1 for Orin and get the Ubuntu 20.04 system and CUDA 11.4. There is no problem to manually compile the source code of ROS2 Humble in the Ubuntu 20.04, but when manually compiling the source code of Autoware. Universe, it will be prompted that the library is missing. Note that you cannot use rosdep to install the missing dependent library at this time, but manually search for the source code of the dependent library. When downloading, the ROS2 Humble version is preferred, followed by the ROS2 version. After all the missing dependency libraries are compiled and installed, there are still individual packages in the Autoware. Universe that cannot be compiled, such as those related to TVM and yabloc. These packages do not affect the operation of the core link. At present, we have completed the verification on the real vehicle according to the above method.