autowarefoundation/autoware_universe #1930: `obstacle_avoidance_planner` sometimes crash when the vehicle goes beyond goal point

VRichardJP, 2022-09-21:
### Checklist

- [X] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [X] I've searched other issues and no duplicate issues were found.
- [X] I'm convinced that this is not my fault but a bug.

### Description

Several time today, the obstacle_avoidance_planner crashed in my vehicle. With GDB, I identified that `full_traj_points` below trajectory is sometimes empty:
https://github.com/autowarefoundation/autoware.universe/blob/e2c5eda7524ff11bab791f9c6ac19c884a3c314d/planning/obstacle_avoidance_planner/src/node.cpp#L1355

I modified the code like so, and the node stopped crashing:
```
  const auto fine_traj_points = generateFineTrajectoryPoints(path_points, full_traj_points);
  if (fine_traj_points.empty()) {
    RCLCPP_ERROR(get_logger(), "FIXME: fine_traj_points.empty()");
    trajectory_points = points_utils::convertToTrajectoryPoints(path_points);
    return trajectory_points;
  }
```

(however I have no idea if it is the correct way to handle that.)

### Expected behavior

No crash

### Actual behavior

Crash

### Steps to reproduce

Seems to happen when the vehicle goes beyond goal point (in manual driving)

takayuki5168, 2022-09-22:
@VRichardJP Thank you so much for your bug report and suggestion for bug fix.
I agree with your suggestion. Could you make a PR?

VRichardJP, 2022-09-22:
I would be happy to contribute. But is this workaround the good way to handle that situation?

For example a special case for empty trajectories could be added to `generateFineTrajectoryPoints()` instead.

takayuki5168, 2022-09-23:
@VRichardJP Hmm, I'm not 100% sure, but check-and-validate as follows looks better to me.

```cpp
 const auto full_traj_points =
    points_utils::concatTrajectory(optimized_traj_points, extended_traj_points);

  if (full_traj_points.empty()) {
    trajectory_points = points_utils::convertToTrajectoryPoints(path_points);
    return trajectory_points;
  }
```

xmfcx, 2022-12-13:
@VRichardJP could you create a PR for this with your changes? So we can try to address your problems there.

VRichardJP, 2022-12-15:
The vehicle I had the problem with has been updated with a newer version of the code since then. Even though I did not keep my change after the update, I have never faced the problem again. So I guess it has been fixed somehow?

closed 2022-12-15 by VRichardJP