autowarefoundation/autoware_universe #1939: RTC Auto mode manager keeps "Waiting for service... [blind_splot]"

VRichardJP, 2022-09-22:
### Checklist

- [X] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [X] I've searched other issues and no duplicate issues were found.
- [X] I'm convinced that this is not my fault but a bug.

### Description

The RTC auto mode manager is spamming the following message on my machine:
```
[planning.scenario_planning.lane_driving.behavior_planning.rtc_auto_mode_manager]: Waiting for service... [blind_spot]
```

I am using pretty much the default configuration for planning modules. What does this message imply? is the module blind_spot not loaded properly? is some interfacing missing? does it mean it is unsafe to drive in autonomous mode?

### Expected behavior

RTC manager working without error.

### Actual behavior

RTC is stuck waiting for something.

### Steps to reproduce

?

takayuki5168, 2022-09-22:
@rej55 Could you check this

VRichardJP, 2022-09-24:
@rej55 which module is supposed to start the services `*/internal/blind_spot`? I can't see any `*/internal/*` service in the code base.

rej55, 2022-09-26:
@VRichardJP 
Sorry for late reply.
The message is sent from following.

https://github.com/autowarefoundation/autoware.universe/blob/17e16d77e9f4f5f8762059d488137de762d28ef2/planning/rtc_auto_mode_manager/src/rtc_auto_mode_manager_interface.cpp#L26-L37

In your environment, do you run blind_spot module in behavior_velocity_planner?
If not, you should delete blind_spot from config file.

https://github.com/autowarefoundation/autoware.universe/blob/main/planning/rtc_auto_mode_manager/config/rtc_auto_mode_manager.param.yaml

VRichardJP, 2022-09-26:
You're right!
I thought it was running but it was disabled in the behavior_velocity_planner config file

closed 2022-09-26 by VRichardJP