autowarefoundation/autoware_universe #1956: `obstacle_avoidance_planner::generateOptimizedTrajectory` sometimes produce NaN filled trajectories

VRichardJP, 2022-09-26:
### Checklist

- [X] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [X] I've searched other issues and no duplicate issues were found.
- [X] I'm convinced that this is not my fault but a bug.

### Description

While debugging `obstacle_avoidance_planner`, I observed that the output of `obstacle_avoidance_planner::generateOptimizedTrajectory` contains sometimes NaN values:
https://github.com/autowarefoundation/autoware.universe/blob/dd190a35f73c9ddd2b9b3a2533add63ae705ab85/planning/obstacle_avoidance_planner/src/node.cpp#L921

I couldn't find what exactly produce the NaNs, but when it does, the module crashes a bit further because of some data validation deeper down:
https://github.com/autowarefoundation/autoware.universe/blob/dd190a35f73c9ddd2b9b3a2533add63ae705ab85/planning/obstacle_avoidance_planner/src/node.cpp#L1368

I first thought I could easily get rid of that issue by simply removing the NaN values, but actually the generated trajectories are filleds with NaNs, and no point is left after the process, leading the planner not to publish any path (or publishing empty ones, but in any case, the vehicle does not seem to generate any new path and stop at the end of the old one.)

When this happens, I also tried to simply call `resetPrevOptimization` to force the module to re-generate the trajectory from scratch. But it always end up generating NaN-filles trajectories.

I don't have any rosbag or lanelet to provide. But on my vehicle, this problem would appear very easily with the default configuration (maybe just with `enable_obstacle_avoidance=true`). 
I am not sure, but it seems to happen most in intersections, when the vehicle has to turn.

### Expected behavior

No crash, No NaNs

### Actual behavior

NaNs and crash

### Steps to reproduce

?

### Versions

- Ubuntu 20.04
- ROS2 galactic
- branch: main

takayuki5168, 2022-09-26:
@VRichardJP 
With our maps in TIER IV, we don't have this kind of issue. I guess it may be because of your map.
I'm sorry, but I cannot help you without your map.

VRichardJP, 2022-09-26:
I totally get it. I am discussing with the site owner whether I can share the lanelet/pcd map with you.

VRichardJP, 2022-09-27:
You can use this lanelet map:
[lanelet2_map.txt](https://github.com/autowarefoundation/autoware.universe/files/9651142/lanelet2_map.txt)

The problem mostly occured at the 2 marked location, when the vehicle is turning left at the intersection:
[image]

takayuki5168, 2022-09-27:
@VRichardJP Let me check first if you've checked if the same issue occurs in the simulation?

takayuki5168, 2022-09-27:
In my environment, the issue didn't occur.

VRichardJP, 2022-09-27:
I have not checked the simulation. Because I had to quickly fix the AD I didn't take the time to record any rosbag :(

takayuki5168, 2022-11-27:
@VRichardJP Can I close the issue for now since there is no reproduction method and not reported by others. We can reopen if the problem occurred again.
Or do you any reason to keep this issue open?

VRichardJP, 2022-11-27:
You can close it.
I have not faced the problem for a while either. Maybe it was just my old setup which was broken.

closed 2022-11-28 by takayuki5168