autowarefoundation/autoware_universe #1958: `obstacle_avoidance_planner` randomly generate zigzagging path while on a straight line

VRichardJP, 2022-09-26:
### Checklist

- [X] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [X] I've searched other issues and no duplicate issues were found.
- [X] I'm convinced that this is not my fault but a bug.

### Description

Sometimes, when my vehicle is on a straight line (never in curves or slight curves), the `obstacle_avoidance_planner` generates a new path with small zigzags instead of a straight path.

This is very annoying, because it makes the car suddenly brake, for no apparent reason (no wall is shown on rviz). It does not seem there is anything particular on the road either. And in my case, the lanelet is quite large.

Here is an example:
- normal path
[image]

- sudden zigzag:
[image]

I know for sure it is from `obstacle_avoidance_planner`, because when I set `option.skip_optimization` to true, the problem never appear again.

### Expected behavior

`obstacle_avoidance_planner` should not zigzag

### Actual behavior

sometimes zigzag

### Steps to reproduce

1. straight lanelet
2. enable avoidance?

takayuki5168, 2022-09-26:
@VRichardJP 
Could you visualize "drivable area" on Rviz? There is drivable area rviz plugin in the tier4_planning_rviz_plugin package. This package is contained in autoware.universe, and you just select the path from behavior_velocity_planner as drivable area plugin on Rviz.

The obstacle avoidance planner tries to make trajectory footprints inside the drivable area. I guess the drivable area may be something wrong.

VRichardJP, 2022-09-26:
I checked several times the drivable area. When the zigzag appear, it does not seem like anything is wrong (like there is something in the middle of the path.) However I am not sure at 100%. 

When I first started my test on this site, I was using a 10 km/h speed limit, and there was absolutely no issue. The problem started to appear when I used higher speed (20~40 km/h). I may have changed many things in the meantime I am not sure at 100% it would still have no problem at low speed.

I looked many times at the pointcloud data and tried to observe if any point would be mistakenly detected as obstacle, but I did not see any.

Unfortunately, I left that site and won't be able to test again.

takayuki5168, 2022-09-26:
@VRichardJP Currently, obstacle avoidance planner does not consider obstacles. Just make footprints inside the drivable area generated by behavior_path_planner.

Therefore, it is often easy to reproduce what happened with the physical vehicle in the simulator regarding to the obstacle avoidance planner.

Could you give me the map where the problem happens or rosbag including behavior_velocity_planner's output?

VRichardJP, 2022-09-27:
You can use this lanelet2 map:
[lanelet2_map.txt](https://github.com/autowarefoundation/autoware.universe/files/9651140/lanelet2_map.txt)

The problem appears only on straight roads, and basically all of them:
[image]

I mostly tested the anti-clockwise loop, but I guess the zigzagging may also appear on the clockwise loop.

takayuki5168, 2022-09-27:
@VRichardJP Let me check first if you've checked if the same issue occurs in the simulation?

takayuki5168, 2022-09-27:
I checked some cases you mentioned, but nothing weird occurred.
[image]

takayuki5168, 2022-09-27:
It seems lanelet map has no issues.
Obstacle avoidance lanner mostly depends on the lanelet (more precisely, drivable area generated by behavior_path_planner) since it does not see any dynamic obstacles.

A year before, there was a bug of drivable area when the ego pose precision was not so good, causing the trajectory from obstacle avoidance planner not alinged to the lane.
Like this, One of the dynamic something affecting obstacle avoidance planner is ego pose.

I'm sorry, but I cannot comment anymore with the current information.
Did you get a rosbag when the issue occurred? If so, I want to check it.

takayuki5168, 2022-09-27:
Or did you change the parameter about obstacle_avoidance_planner and drivable area (in behavior_path_planner)

takayuki5168, 2022-09-27:
Or are you using a large vehicle (like a bus) compared to sample_vehicle?

VRichardJP, 2022-09-27:
@takayuki5168 

I never tested the lanelet in simulation.

As far as planner is concerned, I was using mostly the default configuration. I only changed irrelevant values such as max_accel.

The vehicle is a BYD 6, so a normal car size.

I understand it could come from shaky localization. From the naked eye it does not seem the position ever jumped, even slightly. The localization also never got lost.

That said the site itself was not the best for NDT (many trees, almost no building). The Twist data is also not that precise (longitudinal speed has 1km/h step increment).

I have tested EKF with RTK-GNSS and IMU acceleration, but I would still get the zigzag issue (with more or less the same frequency).

Unfortunately, I am not on site anymore (and won't come back) so it will be difficult to get any data.

I am still wondering why the issue never occured in curves, but always on straight lanes. By looking at the configuration, I see the MPT has different weight parameters for straight road and curved ones. Could MPT be the cause of my issue?

takayuki5168, 2022-09-27:
I got it. When the weird trajectory is generated next time, it would be really helpful if you could record a rosbag and share it with me.

> By looking at the configuration, I see the MPT has different weight parameters for straight road and curved ones. 

No. There are no parameters depending on whether the road is straight or not.
There are two optimization in obstacle avoidance planner
- Elastic Band: to smooth the path
- MPT: to make footprints inside the drivable area

> Could MPT be the cause of my issue?

If the vehicle is large against the road width, MPT may be the cause, but it seems it's not.

I cannot analyze more with the current information. Rosbag or simulation reproduction is required at least.
As I said, my guess is the input of obstacle_avoidance_planner (e.g. drivable area) is something wrong.

xmfcx, 2022-10-04:
@VRichardJP can you share some data to reproduce this issue?

VRichardJP, 2022-10-05:
@xmfcx Unfortunately I don't have anything but the map and a few screenshots/videos. I won't be able to go to have access to the vehicle anytime soon.

mitsudome-r, 2022-10-25:
@VRichardJP Is it okay if I close the issue for now? You can reopen it whenever you were able to reproduce the issue. Otherwise, we can't do anything at the moment.

VRichardJP, 2022-10-25:
Yes no problem

closed 2022-10-25 by VRichardJP