autowarefoundation/autoware_universe #1959: `signal_processing` tests rely on uninitialized variable

ambroise-arm, 2022-09-27:
### Checklist

- [X] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [X] I've searched other issues and no duplicate issues were found.
- [X] I'm convinced that this is not my fault but a bug.

### Description

The low pass filter unit test of the `signal_processing` package fails often on Arm platforms.

### Expected behavior

The unit test passes consistently.
There are no uninitialized variables.

### Actual behavior

I see a failure rate of the unit test above 50%.

Example failure output:
```
1: [ RUN      ] lowpass_filter_twist.filter
1: /home/user/autoware/src/universe/autoware.universe/common/signal_processing/test/src/lowpass_filter_test.cpp:44: Failure
1: The difference between filtered_twist->linear.y and 0.0 is 7.1114927535527936e+197, which exceeds epsilon, where
1: filtered_twist->linear.y evaluates to -7.1114927535527936e+197,
1: 0.0 evaluates to 0, and
1: epsilon evaluates to 9.9999999999999995e-07.
1: [  FAILED  ] lowpass_filter_twist.filter (0 ms)
1: [----------] 1 test from lowpass_filter_twist (0 ms total)
```
---

Additionally, when building the package with `--cmake-args -DCMAKE_BUILD_TYPE=Debug`, the unit test fails consistently on both x86 and arm with:
```
1: [ RUN      ] lowpass_filter_1d.filter
1: test_signal_processing: /usr/include/boost/optional/optional.hpp:1207: boost::optional<T>::reference_type boost::optional<T>::get() [with T = double; boost::optional<T>::reference_type = double&]: Assertion `this->is_initialized()' failed.
1: -- run_test.py: return code -6
```
Which highlights that the package relies on uninitialized `optional` variable.

### Steps to reproduce

Run `colcon test --packages-select signal_processing` on an arm64 platform a few times in a row. 

### Possible causes

The `LowpassFilterTwist` class has a `x_` member variable of type `boost::optional<geometry_msgs::msg::Twist>`, which is not initialized. And the unit test expects the unitialized value to evaluate 0, which is not always the case.
Interestingly, it seems to consistently pass on x86.

### Additional context

Simply default-initializing the variable in question is not enough as the unit test expects it to be uninitialized afterwards.
Packages that make use of this would need to be checked to see what they rely on and what they expect.

**Question**: because the issue is shown explicitly when compiling with `Debug`, would it make sense to have that in CI in addition to the currently-tested `Release`?

ambroise-arm, 2022-09-27:
cc @takayuki5168 as package maintainer
cc @kenji-miyake for the CI question

takayuki5168, 2022-09-27:
@ambroise-arm 
Thank you for your report.

Sorry, this is my fault. I created a PR, so could you check please.
https://github.com/autowarefoundation/autoware.universe/pull/1960

kenji-miyake, 2022-09-27:
@ambroise-arm Thank you for your detail report.

> Question: because the issue is shown explicitly when compiling with Debug, would it make sense to have that in CI in addition to the currently-tested Release?

Do you think we can use `RelWithDebInfo` instead? I'm concerned that `Debug` will be very slow.
https://github.com/autowarefoundation/autoware-github-actions/blob/c3e0ce4316d83468a9922bbdc62d85a6a80bc5a5/colcon-build/action.yaml#L65

ambroise-arm, 2022-09-27:
@kenji-miyake Only `Debug` enables the DEBUG flag of interest here. In my head the debug build would be faster than the release build because there are less optimizations that the compiler needs to do. I will check that.

kenji-miyake, 2022-09-27:
I was worried about the execution time of tests.
But considering there aren't so many tests right now, I'll use `Debug`. Thanks!

kenji-miyake, 2022-09-28:
@ambroise-arm Please let me confirm.
Did this occur on Humble? The `galactic` CI with `Debug` has passed.

https://github.com/autowarefoundation/autoware.universe/actions/runs/3137771803/jobs/5096330982
[image]

ambroise-arm, 2022-09-28:
@kenji-miyake Very weird. The failure on the `(humble, Debug)` configuration matches the one I was seeing locally. But I was using galactic. I have re-run the tests locally but using the CI setup (with the mixins and the coverage enabled) and I still see the same failures. I don't understand how the `(galactic, Debug)` configuration passes here.

kenji-miyake, 2022-09-29:
@ambroise-arm I was able to reproduce the issue. But it seems `Debug` uses much more storage that `Release`. We have to cut down on the usage also for GitHub-hosted runners.
https://github.com/autowarefoundation/autoware.universe/actions/runs/3143374868

[image]

takayuki5168, 2022-09-29:
FYI. I merged the fix PR for this issue. However, you keep discussing the issue, so I don't close this issue.
https://github.com/autowarefoundation/autoware.universe/pull/1960

closed 2022-10-04 by xmfcx