autowarefoundation/autoware_universe #1988: Investigate `lidar_apollo_segmentation_tvm` detection quality

ambroise-arm, 2022-09-29:
### Checklist

- [X] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [X] I've searched other issues and no duplicate issues were found.
- [X] I've agreed with the maintainers that I can plan this task.

### Description

I have tested the TVM package replacing the TRT package in the [rosbag demo](https://autowarefoundation.github.io/autoware-documentation/main/tutorials/ad-hoc-simulation/rosbag-replay-simulation/) as part of the PR: https://github.com/autowarefoundation/autoware.universe/pull/1181#issuecomment-1259269257. And the quality of the detection wasn't as good as the detection from the TRT package, at least visually. Some cars were not detected, the lines and circles which indicate the velocity (or planned trajectory, I'm not sure) were wrong, at least on parked vehicles. And the type of detected vehicles might be wrong, it would need to be checked.

### Purpose

Get the detection quality of the TVM package on par with the TRT package.

### Possible approaches

There are multiple possible causes for the difference:

- The TVM version of the package is derived from an [old version from TierIV](https://github.com/tier4/lidar_instance_segmentation_tvm); I expect the TRT version in this repository is more recent
- The neural network may have been updated since then; trained differently/better
- The port may have introduced issues/bugs
- The parameters may not be correct or not all there

### Definition of done

- [ ] confirm and document the issue
- [ ] find the cause(s)
- [ ] update the `lidar_apollo_segmentation_tvm*` packages and/or the [neural network](https://github.com/autowarefoundation/modelzoo/tree/master/perception/lidar_obstacle_detection/baidu_cnn/onnx_bcnn) in use, depending on what is necessary

angry-crab, 2022-10-17:
[image]
Issue confirmed. There are quite a lot of false negative detections.

angry-crab, 2022-10-17:
According to @yukkysaito san, the weights have been update. I'll update the `modelzoo` repo.

angry-crab, 2022-10-18:
Converting to `onnx` format seems better since `caffe` operations are not fully supported and will probably be removed in the near future. 
https://discuss.tvm.apache.org/t/caffe2-frontend-support-is-being-dropped-to-unblock-pytorch-update/12442

ambroise-arm, 2022-10-18:
@angry-crab I don't think the Autoware pipeline for tvm only supports onnx, and other formats could be used. It is really down to what TVM supports. If caffe support is dropped in TVM, then I agree that it can't be used as is and needs to be converted.

ambroise-arm, 2022-10-26:
I couldn't convert the current caffemodel files with the microsoft tool, it complains about the "Slice" layer.
I've also tried to use small tools like https://github.com/asiryan/caffe2onnx, which clains to support "Slice", but there is a "Silence" layer in the network that isn't supported.

angry-crab, 2022-10-31:
> I couldn't convert the current caffemodel files with the microsoft tool, it complains about the "Slice" layer. I've also tried to use small tools like https://github.com/asiryan/caffe2onnx, which clains to support "Slice", but there is a "Silence" layer in the network that isn't supported.

[image]

I got it worked after using the converted model. It looks good. `caffemodel` file contains weights while `prototxt` file describes the structure of network, which does not have `Slice` and `Silence` layers.

PR #2185 (2022-11-15): fix(lidar_apollo_segmentation_tvm, tvm_utility): fixed output dim and pipeline

## Description

Resolve #1988 
Changed default allocation device according to config. Fixed output data transfer in case of GPU involved. 
Reason discussed in the follwing issue.
#1898 

## Pre-review checklist for the PR author

The PR author **must** check the checkboxes below when creating the PR.

- [x] I've confirmed the [contribution guidelines].
- [x] The PR follows the [pull request guidelines].

## In-review checklist for the PR reviewers

The PR reviewers **must** check the checkboxes below before approva [... cut]
closed 2022-11-15 by xmfcx