autowarefoundation/autoware_universe #2047: Trajectory follower: Memory leak

smorita-esol, 2022-10-11:
### Checklist

- [X] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [X] I've searched other issues and no duplicate issues were found.
- [X] I'm convinced that this is not my fault but a bug.

### Description

Trajectory follower seems to have memory leaks when it runs as a node of The Planning Simulator Demo below.  
```
https://autowarefoundation.github.io/autoware-documentation/main/tutorials/ad-hoc-simulation/planning-simulation/  
```

### Expected behavior

Memory consumption(RSS) of the process to which trajectory follower node belongs falls within a certain amount.

### Actual behavior

Memory consumption(RSS) of the process to which trajectory follower node belongs falls within a certain amount.
Actual behavior:
The memory consumption of the process continuously increases up to over 1GBytes as below.  
[image]

### Steps to reproduce

1. Follow the instructions of the Planning Simulator Demo above, using the docker image below.
    ```
    humble-20220915-amd64:           00e518e86ad3f9a38378f6ab28db48fecce69b75
    ```  
    To measure the memory consumption only of the process to which the trajectory follower belongs, you should modify the launch file to contain the trajectory follower node independently in a component container.
2. Launch the Planning simulator demo.
3. Proceed the steps till "4. Set a goal pose for the ego vehicle"    
    Set the vehicle on an arbitrary position as below.
  
[image]

 
4. Monitor the memory consumption of the separated component container.

### Versions

OS: Ubuntu 22.04 + Docker environment  
ROS2: Humble included in the docker image   humble-20220915-amd64  
Autoware: the one in the docker image above.  

### Possible causes

It seems to be caused by some C++ code of the trajectory follower.  

### Additional context

None

VRichardJP, 2022-10-13:
Sorry to jump in.
By looking at [`csc_matrix` implementation](https://github.com/osqp/osqp/blob/master/src/cs.c#L12), the memory leak seems simple to explain from your valgrind logs:
- `csc_matrix()` allocate new memory, which should be freed by the user
- in [`initializeProblem()`](https://github.com/autowarefoundation/autoware.universe/blob/f234a7d9ae044e0f53a3679060dad08aab5ec90e/common/osqp_interface/src/osqp_interface.cpp#L264), the `csc_matrix()` function is called twice to overwrite `m_data->P` and `m_data->A` pointers, so any old data that was previously held by those pointers is now lost.

maxime-clem, 2022-10-13:
Thanks a lot @VRichardJP for finding the cause of the issue.
I will prepare a PR with the fix you suggested.

closed 2022-10-18 by maxime-clem