autowarefoundation/autoware_universe #2048: [map_height_fitter] Improve height estimation logic

VRichardJP, 2022-10-11:
### Checklist

- [X] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [X] I've searched other issues and no duplicate issues were found.
- [X] I've agreed with the maintainers that I can plan this task.

### Description

With the current implementation of [`get_ground_height`](https://github.com/autowarefoundation/autoware.universe/blob/4e080d2c05da5b1a0356c1ec16ae3c77676914f7/localization/pose_initializer/src/map_height_fitter/map_height_fitter_core.cpp#L46) When the user sets the initial pose using Rviz, if the initial pose is not within 1 meter of any map point, the `map_height_fitter` node will fail to estimate its height. Most likely, the initial localization will be wrong.

That situation typically occurs when the vehicle is located in some garage or building, whose inside has not mapped. Like here:
 
[image]

[image]

It is also likely the issue would occur if the 3D map has no ground (I don't know if it is a thing)

### Purpose

Improve usability of the height estimator.

### Possible approaches

Obviously, the hardcoded 1.0 meter radius search makes the implementation not very flexible. 

I think that simply increasing the radius search would have some undesirable side effects.

One simple improvement would be to consider all point within r+1.0 radius, where r is the distance to the closest point. This way we make sure we always find a bunch of points, yet we don't look unecessarily far.

The only downside I see is that it takes slightly more time, since we explore the map twice (once to find the closest point, another time to find min height within radius r+1.0)

Any other suggestion?

### Definition of done

Pose height estimation give a "decent" estimation even when there is no map point very close.

YamatoAndo, 2022-10-11:
> One simple improvement would be to consider all point within r+1.0 radius, where r is the distance to the closest point. This way we make sure we always find a bunch of points, yet we don't look unecessarily far.

I think it is a good idea!

YamatoAndo, 2022-10-11:
> The only downside I see is that it takes slightly more time, since we explore the map twice (once to find the closest point, another time to find min height within radius r+1.0)

I think it is not problem if we explore the map twice, since dynamic map loading will be implemented soon and the area where the map is loaded will be smaller than before.
https://github.com/orgs/autowarefoundation/discussions/2812

isamu-takagi, 2022-10-12:
@YamatoAndo @VRichardJP 
I think it is unreliable if there is no ground under the vehicle area. Here is an example. If the ground is sloped or stepped, the radius calculated from the closest point is not helpful. Moreover, the inside of the garage given as an example will give erroneous results if there is a ceiling when measured from the outside. Therefore, initialization results cannot be guaranteed at locations where the ground is not measured.

[image]

VRichardJP, 2022-10-12:
I totally understand the proposed change is not perfect, but I still think it is better than the current implementation. 

In particular, I don't think there is a case where it perform worse than the current implementation. Typically, for both examples you provided, there is no point within a 1 meter radius, so the returned height would be whatever Z value is currently stored in the input pose (most likely wrong?)

isamu-takagi, 2022-10-12:
Ah, I see. Both the old and new logic fail in these cases, but the new one improves some cases.

closed 2022-10-25 by VRichardJP