autowarefoundation/autoware_universe #2052: Gear should not be changed while autoware is not engaged

VRichardJP, 2022-10-12:
### Checklist

- [X] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [X] I've searched other issues and no duplicate issues were found.
- [X] I'm convinced that this is not my fault but a bug.

### Description

I am using a vehicle with pacmod3.

When the vehicle is engaged (via /control/control_mode_request) but autoware is not, if the user sets a new goal point, the vehicle gear is automatically changed to DRIVE after the planning is completed.

Considering autoware is still not engaged at this point, it seems strange for autoware to change the gear.

Eventually, it could make sense to have the gear changed to PARK while autoware is not engaged (the same way a brake command is sent as long as autoware is not engaged.)

### Expected behavior

Autoware should not change gear to DRIVE while not engaged.

### Actual behavior

Gear is changed to DRIVE after planning is finished

### Steps to reproduce

1. Set goal point without engaging autoware
2. check /control/command/gear_cmd topic

mitsudome-r, 2022-10-25:
I think whether this is a bug or not depends on what /autoware/engage means.
@VRichardJP Just to confirm do you have any issues with changing gear to DRIVE before engaging?

VRichardJP, 2022-10-26:
There is no bug/problem with the current behavior, it is just weird from the user poV. Autoware engage has been split into "autoware engage" and "vehicle engage". So one would expect the vehicle does not follow any autoware command (including gear change) as long as autoware is not engaged. 

When autoware is not engaged, I think the vehicle behavior should be:
- push the brake (current behavior)
- keep the current gear (not current behavior)

xmfcx, 2022-12-13:
We should add this to documentation or link existing documentation here.

brkay54, 2022-12-21:
@VRichardJP I agree with you, created PR to send current gear if autoware is not engaged. Please check the PR, let me know if it solves your problem.

VRichardJP, 2022-12-24:
I see you subscribe to the current command and send it back when the vehicle is not engaged. 

Even though it does exactly the same, what if we don't simply send `GearCommand::NONE`? If `NONE` is a valid value, we can understand it as "no command". In other words "do not change gear".

Then of course, it is up to the vehicle interface to interpret `NONE` value correctly.

brkay54, 2022-12-24:
After thinking about it, I am not sure about sending current gear if vehicle in Autonomous Mode and autoware is not engaged. I think, if vehicle's gate mode is ::AUTO and autoware is not engaged, vehicle_cmd_gate should send GearCommand::PARK.

For the situation you mentioned, vehicle_cmd_gate is listening the commands from autoware because current gate is AUTO. If current gate is auto, autoware should send the cmds if it is engaged. But if it is not engaged, we should send proper cmds  until autoware is engaged. In this time (i think) vehicle's gear should be PARK in AUTO gate until autoware engaged, it will be more safely.

VRichardJP, 2022-12-25:
From the user point-of-view, I think both options are fine:
- send brake only (no gear change)
- or send brake + parking gear

The brake + park gear feels like the safer option. 

However, I am curious what happens if both autoware and the vehicle are engaged almost at the same time. For example，if vehicle engage is processed slightly before the autoware engage, the gear may change to park, then to drive right after. 

But maybe it is something that can be handled on vehicle driver side. Maybe simultaneous engage is not really a thing either.

Up to you 😁

brkay54, 2023-01-10:
Thank you for your suggestions. Sending park gear may be dangerous in some situations such as disengaging autoware while driving, It will change gear to Park while vehicle is moving. So, I implemented the first option. Please check, if this implementation solves the problem, then we can close the issue.

closed 2023-01-10 by brkay54