autowarefoundation/autoware_universe #2072: map_loader fails to load lanelet map and another map is loaded

VRichardJP, 2022-10-14:
### Checklist

- [X] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [X] I've searched other issues and no duplicate issues were found.
- [X] I'm convinced that this is not my fault but a bug.

### Description

I am sorry about the confusing title, as I am myself confused over the following case:

I start autoware planning simulator with:
```
$ ros2 launch autoware_launch planning_simulator.launch.xml map_path:=/data/autoware_data_universe/xxx
```
In the first few lines of logs, I can see the lanelet2_map_loader has crashed with the message:

```
[ERROR] [launch_ros.actions.load_composable_nodes]: Failed to load node 'lanelet2_map_loader' of type 'Lanelet2MapLoaderNode' in container '/map/map_container': Component constructor threw an exception: Could not find lanelet map under /data/autoware_data_universe/xxx/lanelet2_map.osm
```

Suprisingly the file exist:
```
$ ll /data/autoware_data_universe/xxx/lanelet2_map.osm
-rw-rw-r-- 1 sig sig 114755 Jul 27  2021 /data/autoware_data_universe/xxx/lanelet2_map.osm
```

More surprisingly, I see in Rviz that another lanelet and pcd have been loaded! (located in /data/autoware_data_universe/yyy)

[image]

Just to double check, this is what my xxx map looks like:
[image]

### Expected behavior

The correct map is loaded

### Actual behavior

Magic

### Steps to reproduce

?

kminoda, 2022-10-14:
This is just a quick thought, but maybe killing all the ROS processes may solve the issue.
```
pkill ros2
pkill rviz2
pkill aggregator_node
ps aux | grep python3 | grep ros2 | grep -v grep | awk '{ print "kill ", $2 }' | sh
ps aux | grep python3 | grep rqt_reconfigure | grep -v grep | awk '{ print "kill ", $2 }' | sh
ps aux | grep component_container | grep -v grep | awk '{ print "kill ", $2 }' | sh
ps aux | grep robot_state_publisher | grep -v grep | awk '{ print "kill ", $2 }' | sh
ps aux | grep topic_tools/relay | grep -v grep | awk '{ print "kill ", $2 }' | sh
ps aux | grep "ros-args" | grep -v grep | awk '{ print "kill ", $2 }' | sh
```

VRichardJP, 2022-10-14:
You are right, I can see many nodes in `ros2 node list`. There are not because of some zombie process, but because there is another instance of autoware running on the same network. Next time I will be careful and start ROS2 with localhost only ;)

closed 2022-10-14 by VRichardJP