autowarefoundation/autoware_universe #2079: Vehicle not able to avoid static obstacle by changing lane.

VRichardJP, 2022-10-17:
### Checklist

- [X] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [X] I've searched other issues and no duplicate issues were found.
- [X] I'm convinced that this is not my fault but a bug.

### Description

I am testing static obstacle avoidance with the planning simulator.

I have set `enable_avoidance_over_same_direction` to `true` so that the vehicle is able to avoid static obstacles when there is not enough space on the current lane.

However, when I try it with a fake bus on the shoulder of the road, the vehicle seems not to consider it is possible to leave the lane:
[image]

And yet, if I move the goal point to the other lane, autoware is able to generate change lane:
[image]

If autoware can change lane, surely it should be able to avoid the bus? Is there any other parameter besides `enable_avoidance_over_same_direction` I should consider changing?

### Expected behavior

Ego vehicle can avoid the bus in the situation above

### Actual behavior

The vehicle is stuck behind the bus for ever.

### Steps to reproduce

1. use map with 2 lanes
2. add bus on the shoulder of the road, so that there is not enough space to avoid it by staying on the lane

zulfaqar-azmi-t4, 2022-10-17:
Avoidance Module's page describe a few way to debug this.
https://autowarefoundation.github.io/autoware.universe/main/planning/behavior_path_planner/behavior_path_planner_avoidance-design/#how-to-debug

May I ask your help to share with us the `drivable_area_boundary` and the visualization marker screen shot?

VRichardJP, 2022-10-17:
Thank you for the link,

I have added a few markers from `/planning/scenario_planning/lane_driving/behavior_planning/behavior_path_planner/debug` with  `publish_debug_marker` set to `true`. But I don't see much information:

[image]

zulfaqar-azmi-t4, 2022-10-17:
That indeed looks weird. The lane boundary search seems to look fine.

Does this happens in the whole map or only at the particular area?
Can you run 

```
ros2 topic echo /planning/scenario_planning/lane_driving/behavior_planning/behavior_path_planner/debug/avoidance_debug_message_array
```
And share the printed results?

VRichardJP, 2022-10-17:
The output is empty :(
```
---
header:
  stamp:
    sec: 1665976208
    nanosec: 483875373
  frame_id: ''
avoidance_info: []
---
```

zulfaqar-azmi-t4, 2022-10-17:
I've checked the latest commit and everything seems to work fine.

I would suggest the following

1. Set the param `publish_debug_marker` to `true` (you might have already done this, but just in case)
```
ros2 param set /planning/scenario_planning/lane_driving/behavior_planning/behavior_path_planner avoidance.publish_debug_marker true
```

2. then add the following marker to see if the avoidance is currently running (It's a different topic than the one described in the Readme.

```
/planning/scenario_planning/lane_driving/behavior_planning/behavior_path_planner/debug/avoidance
```

Try to move delete and re-add the object to see if there is any changes.

3. If there are also nothing, please run 
```
ros2 topic echo /api/external/get/rtc_status 
```
In the printed rtc result, for avoidance, the module type should be either 3 or 4
```
  module:
    type: 3
```
If it is not 3 or 4, then please share to us the output

4. Just in case, please check whether there is error by running
```
ros2 launch autoware_launch planning_simulator.launch.xml map_path:=$HOME/autoware_map/sample-map-planning vehicle_model:=sample_vehicle sensor_model:=sample_sensor_kit | grep "ERROR"
```

```
ros2 launch autoware_launch planning_simulator.launch.xml map_path:=$HOME/autoware_map/sample-map-planning vehicle_model:=sample_vehicle sensor_model:=sample_sensor_kit | grep "died"
```

5. If the above still yield nothing, we might need the map and the position where this happens in the map, and please let us know what commit that you're currently in by running the following commands in the `autoware.universe` folder.
```
git rev-parse --short HEAD

```

VRichardJP, 2022-10-17:
I don't see any change :/

And `ros2 topic echo /api/external/get/rtc_status` only output empty arrays.

I also tested with the sample map, but it gives the same result.

My work tree has many changes compared to `main`, but almost none regarding planning modules.
My latest merge is at 4be6931c811422f16d95c1460c046f2540d1f628

Maybe I have something disabled in my config?

zulfaqar-azmi-t4, 2022-10-18:
The avoidance module should be able to be used out of the box.

Just to confirm, can you turn on the drivable area's visualization?
The marker can be add by selecting the following marker

[image]

then add this topic
```
/planning/scenario_planning/lane_driving/behavior_planning/path
```

The drivable area will be shown with the following grey colored cells.

[image]

VRichardJP, 2022-10-18:
I can confirm the drivable area is incorrect on my side:
[image]

but with lane change, there is no problem:
[image]

zulfaqar-azmi-t4, 2022-10-18:
I see. It seems that for some reason the object is not registered as a target object.
Does this also happens for CAR type object?

Can you share your configurations from the `avoidance.param.yaml`?
And confirm that these parameter is configured?

```
      target_object:
        car: true
        truck: true
        bus: true
        trailer: true
        unknown: false
        bicycle: false
        motorcycle: false
        pedestrian: false
```

Also check the 
```
install/behavior_path_planner/share/behavior_path_planner/config/behavior_path_planner_tree.xml
```
to see if avoidance is listed in the behavior tree.

And in the 
```
install/tier4_planning_launch/share/tier4_planning_launch/launch/scenario_planning/lane_driving/behavior_planning/behavior_planning.launch.py
```

VRichardJP, 2022-10-18:
Sorry I won't be able to check that.
I tested autoware universe main branch and there was no issue, so I went through a full merge of the main branch into my work tree. Surprisingly, there were many changes, so maybe I had not been careful last time. In particular, I pulled the behavior_path_planner_tree.xml file, so maybe I broke it on my machine.
I tested again after the merge, and there is now no issue anymore.
Thank you!

closed 2022-10-18 by VRichardJP