autowarefoundation/autoware_universe #2097: Improve behavior_velocity_planner's latency

soblin, 2022-10-18:
### Checklist

- [X] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [X] I've searched other issues and no duplicate issues were found.
- [X] I've agreed with the maintainers that I can plan this task.

### Description

Report from @veqcc.

Currently the latency of `onTrigger` function in behavior_velocity_planner has a big influence in Autoware's entire latency. Below figures show (1) the latency of behavior_path_planner and (2) behavior_velocity_planner respectively, measured in [planning_simulation tutorial](https://autowarefoundation.github.io/autoware-documentation/main/tutorials/ad-hoc-simulation/planning-simulation/).

[image]

[image]

Especially the latency of behavior_velocity_planner is constantly not negligible (about 200ms ) given the behavior_path_planner's publication rate of 10Hz, which indicates its latency is twice the subscription rate.

[image]

@soblin and @veqcc are working to lower this computation time as much as possible. I'm not sure the lower bound of this latency, but it should be at least below the behavior_path_planner's publication rate of 10Hz.

### Purpose

Same.

### Possible approaches

- get some statistics on the functions calls
- refactor current implementation
  -  check the logic flow and return instantly
  - avoid and delay heavy computation
  - cache lanelet stuff because this data is expected to be almost the same in every iteration
- do calculation *in-place*

### Definition of done

2x ~ 3x performance improvement would be good.

soblin, 2022-10-18:
https://github.com/autowarefoundation/autoware.universe/pull/2071

VRichardJP, 2022-10-20:
Which tools do you use for profiling? I have used gperftools in the past but is there any tool recommended for autoware?

Out of curiosity, I went on and tried profiling `behavior_path_planner`/`behavior_velocity_planner`.
I did the following:
- added `prefix="gnome-terminal -- env LD_PRELOAD=/lib/x86_64-linux-gnu/libprofiler.so CPUPROFILESIGNAL=12 CPUPROFILE_FREQUENCY=1000 CPUPROFILE=/tmp/behavior_velocity_planner.profile ",` to `behavior_planning_container` in `behavior_planning.launch.py`
- launched planning_simulator, and prepared a simple environment with a few obstacles.
- started and stopped profiling with `kill -12 <CONTAINER_PID>` (see https://gperftools.github.io/gperftools/cpuprofile.html)
- generated the call tree below with: `google-pprof --web install/behavior_velocity_planner/share/ament_index/resource_index/parent_prefix_path/behavior_velocity_planner /tmp/behavior_velocity_planner.profile` (can also use `--cachegrind` option instead of `--web` to explore the logs with kcachegrind)

[image]

For example, my sample profile shows that most of the computing time is spent in the 4 functions:
- `behavior_path_planner::util::resamplePathWithSpline()`
- `behavior_path_planner::util::imageToOccupancyGrid()`
- `cv::Mat::operator=()` (in `behavior_path_planner::util::generateDrivableArea()`)
- `behavior_path_planner::AvoidanceModule::calcShiftLines()`

Do you have any better way to track down the performance issues?

veqcc, 2022-10-21:
We have used CARET (https://github.com/tier4/caret), which is an open source performance analysis tool developed by TIER IV.
Please read the CARET documentatiion!! (https://tier4.github.io/CARET_doc/main/)

veqcc, 2023-01-19:
The performance improvement result is in the following figure.
At the peaks of the graph, 2~2.5x performance optimizations have been realized.
[image]

closed 2023-01-19 by soblin