autowarefoundation/autoware_universe #2099: planning sometimes fails when using RRT* for parking

TakaHoribe, 2022-10-18:
### Checklist

- [X] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [X] I've searched other issues and no duplicate issues were found.
- [X] I'm convinced that this is not my fault but a bug.

### Description

planning fails when driving in parking lots with `rrtstar` algorithm in the `freespace_planner`.

The solved trajectory by the `freespace_planner` is visualized as in the picture below, but the output from the `motion_velocity_smoother` is not visualized.

[image]

Attach the terminal output for further information.
[image]

### Expected behavior

arrive at a goal with `rrtstar` algorithm in parking scenario.

### Actual behavior

planning fails and it won't arrive at a goal

### Steps to reproduce

1. change the default param in freespace_planner from `astar` to `rrtstar`
2. set a goal to in the parking lot and run in the parking scenario

TakaHoribe, 2022-11-02:
no one is assigned now

shulanbushangshu, 2022-11-10:
I test the "rrtstar" in the parking.The  path display planning is successful, but the trajectory of motion_velocity_smoother display wrong is in the process of control. Simulation of the video as follows:

[image]

https://user-images.githubusercontent.com/102840938/201040744-436d0277-f2e3-4782-9256-b000b19c9f85.mp4

PR #2350 (2022-12-09): fix(freespace_planning_algorithms): fix rrtstar can't arrive goal error

## Description

fix #2099 
## Related links

## Tests performed

**Before**
https://user-images.githubusercontent.com/103234047/203239784-3f031b1c-7935-42ca-bc10-1d120f994fad.mp4

**After**
Vehicle arrived at the goal in all three repetitions. Replanning is triggered in the first trial. The last segment is not traveled because arrive-at-goal tolerance in the second trial.
https://user-images.githubusercontent.com/103234047/203239792-cdc1f242-44ec-4b8e-8c73-4acf9803373b.mp4

## Notes for reviewers

## Pre-review checklist  [... cut]
closed 2022-12-09 by mitsudome-r