autowarefoundation/autoware_universe #2161: Rviz sometimes crash because of RTC

VRichardJP, 2022-10-27:
### Checklist

- [X] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [X] I've searched other issues and no duplicate issues were found.
- [X] I'm convinced that this is not my fault but a bug.

### Description

After a recent code update, I observe that Rviz now crashes when the vehicle reaches some specific places. It seems to be caused by RTCManagerPanel plugin.

Here for example, Rviz always crashes at the same place, a few meters before the intersection:
[image]

The crash also happens when the RTCManagerPanel is not displayed, but the cause is the same.

I caught the backtrace with gdb:
```
[INFO 1666853836.843260057] [rviz2]: Setting goal pose: Frame:map, Position(75661.6, 31122.8, 0), Orientation(0, 0, -0.075138, 0.997173) = Angle: -0.150418 (operator()() at /tmp/binarydeb/ros-galactic-rviz2-8.5.1/src/main.cpp:62)
[INFO 1666854102.374821983] [rviz]: client request (onClickAutowareEngage() at /home/sig/autoware/src/universe/autoware.universe/common/tier4_state_rviz_plugin/src/autoware_state_panel.cpp:295)
[INFO 1666854103.393634553] [rviz]: Status: 1,  (operator()() at /home/sig/autoware/src/universe/autoware.universe/common/tier4_state_rviz_plugin/src/autoware_state_panel.cpp:303)

Thread 1 "rviz2" received signal SIGSEGV, Segmentation fault.
0x00007ffff789dfe4 in QWidget::setStyleSheet(QString const&) () from /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5
(gdb) bt
#0  0x00007ffff789dfe4 in QWidget::setStyleSheet(QString const&) () from /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5
#1  0x00007fff6f7bfdc7 in rviz_plugins::RTCManagerPanel::onRTCStatus(std::shared_ptr<tier4_rtc_msgs::msg::CooperateStatusArray_<std::allocator<void> > const>) () from /home/sig/autoware/install/rtc_manager_rviz_plugin/lib/librtc_manager_rviz_plugin.so
#2  0x00007fff6f7ccec8 in std::_Function_handler<void (std::shared_ptr<tier4_rtc_msgs::msg::CooperateStatusArray_<std::allocator<void> > const>), std::_Bind<void (rviz_plugins::RTCManagerPanel::*(rviz_plugins::RTCManagerPanel*, std::_Placeholder<1>))(std::shared_ptr<tier4_rtc_msgs::msg::CooperateStatusArray_<std::allocator<void> > const>)> >::_M_invoke(std::_Any_data const&, std::shared_ptr<tier4_rtc_msgs::msg::CooperateStatusArray_<std::allocator<void> > const>&&) () from /home/sig/autoware/install/rtc_manager_rviz_plugin/lib/librtc_manager_rviz_plugin.so
#3  0x00007fff6f7ce4c8 in std::__detail::__variant::__gen_vtable_impl<true, std::__detail::__variant::_Multi_array<void (*)(rclcpp::AnySubscriptionCallback<tier4_rtc_msgs::msg::CooperateStatusArray_<std::allocator<void> >, std::allocator<void> >::dispatch(std::shared_ptr<tier4_rtc_msgs::msg::CooperateStatusArray_<std::allocator<void> > >, rclcpp::MessageInfo const&)::{lambda(auto:1&&)#1}&&, std::variant<std::function<void (tier4_rtc_msgs::msg::CooperateStatusArray_<std::allocator<void> > const&)>, std::variant<void (tier4_rtc_msgs::msg::CooperateStatusArray_<std::allocator<void> > const, rclcpp::MessageInfo const&)>, std::variant<void (std::unique_ptr<tier4_rtc_msgs::msg::CooperateStatusArray_<std::allocator<void> >, std::default_delete<tier4_rtc_msgs::msg::CooperateStatusArray_<std::allocator<void> > > >)>, std::variant<void (std::default_delete<tier4_rtc_msgs::msg::CooperateStatusArray_<std::allocator<void> > >, rclcpp::MessageInfo const&)>, std::variant<void (std::shared_ptr<std::function>)>, std::variant<void (std::variant<void (std::default_delete<tier4_rtc_msgs::msg::CooperateStatusArray_<std::allocator<void> > >, rclcpp::MessageInfo const&)>, rclcpp::MessageInfo const&)>, std::variant<void (std::variant<void (std::default_delete<tier4_rtc_msgs::msg::CooperateStatusArray_<std::allocator<void> > >, rclcpp::MessageInfo const&)> const&)>, std::variant<void (std::variant<void (std::default_delete<tier4_rtc_msgs::msg::CooperateStatusArray_<std::allocator<void> > >, rclcpp::MessageInfo const&)> const, rclcpp::MessageInfo const&)>, std::variant<void (std::shared_ptr<tier4_rtc_msgs::msg::CooperateStatusArray_<std::allocator<void> > >)>, std::variant<void (std::shared_ptr<tier4_rtc_msgs::msg::CooperateStatusArray_<std::allocator<void> > >, rclcpp::MessageInfo const&)> >&)>, std::tuple<std::variant<std::function<void (tier4_rtc_msgs::msg::CooperateStatusArray_<std::allocator<void> > const&)>, std::variant<void (tier4_rtc_msgs::msg::CooperateStatusArray_<std::allocator<void> > const, rclcpp::MessageInfo const&)>, std::variant<void (std::unique_ptr<tier4_rtc_msgs::msg::CooperateStatusArray_<std::allocator<void> >, std::default_delete<tier4_rtc_msgs::msg::CooperateStatusArray_<std::allocator<void> > > >)>, std::variant<void (std::default_delete<tier4_rtc_msgs::msg::CooperateStatusArray_<std::allocator<void> > >, rclcpp::MessageInfo const&)>, std::variant<void (std::shared_ptr<std::function>)>, std::variant<void (std::variant<void (std::default_delete<tier4_rtc_msgs::msg::CooperateStatusArray_<std::allocator<void> > >, rclcpp::MessageInfo const&)>, rclcpp::MessageInfo const&)>, std::variant<void (std::variant<void (std::default_delete<tier4_rtc_msgs::msg::CooperateStatusArray_<std::allocator<void> > >, rclcpp::MessageInfo const&)> const&)>, std::variant<void (std::variant<void (std::default_delete<tier4_rtc_msgs::msg::CooperateStatusArray_<std::allocator<void> > >, rclcpp::MessageInfo const&)> const, rclcpp::MessageInfo const&)>, std::variant<void (std::shared_ptr<tier4_rtc_msgs::msg::CooperateStatusArray_<std::allocator<void> > >)>, std::variant<void (std::shared_ptr<tier4_rtc_msgs::msg::CooperateStatusArray_<std::allocator<void> > >, rclcpp::MessageInfo const&)> > >, std::integer_sequence<unsigned long, 4ul> >::__visit_invoke(rclcpp::AnySubscriptionCallback<tier4_rtc_msgs::msg::CooperateStatusArray_<std::allocator<void> >, std::allocator<void> >::dispatch(std::shared_ptr<tier4_rtc_msgs::msg::CooperateStatusArray_<std::allocator<void> > >, rclcpp::MessageInfo const&)::{lambda(auto:1&&)#1}, std::variant<std::function<void (tier4_rtc_msgs::msg::CooperateStatusArray_<std::allocator<void> > const&)>, std::variant<void (tier4_rtc_msgs::msg::CooperateStatusArray_<std::allocator<void> > const, rclcpp::MessageInfo const&)>, std::variant<void (std::unique_ptr<tier4_rtc_msgs::msg::CooperateStatusArray_<std::allocator<void> >, std::default_delete<tier4_rtc_msgs::msg::CooperateStatusArray_<std::allocator<void> > > >)>, std::variant<void (std::default_delete<tier4_rtc_msgs::msg::CooperateStatusArray_<std::allocator<void> > >, rclcpp::MessageInfo const&)>, std::variant<void (std::shared_ptr<std::function>)>, std::variant<void (std::variant<void (std::default_delete<tier4_rtc_msgs::msg::CooperateStatusArray_<std::allocator<void> > >, rclcpp::MessageInfo const&)>, rclcpp::MessageInfo const&)>, std::variant<void (std::variant<void (std::default_delete<tier4_rtc_msgs::msg::CooperateStatusArray_<std::allocator<void> > >, rclcpp::MessageInfo const&)> const&)>, std::variant<void (std::variant<void (std::default_delete<tier4_rtc_msgs::msg::CooperateStatusArray_<std::allocator<void> > >, rclcpp::MessageInfo const&)> const, rclcpp::MessageInfo const&)>, std::variant<void (std::shared_ptr<tier4_rtc_msgs::msg::CooperateStatusArray_<std::allocator<void> > >)>, std::variant<void (std::shared_ptr<tier4_rtc_msgs::msg::CooperateStatusArray_<std::allocator<void> > >, rclcpp::MessageInfo const&)> >) () from /home/sig/autoware/install/rtc_manager_rviz_plugin/lib/librtc_manager_rviz_plugin.so
#4  0x00007fff6f7e4a75 in rclcpp::Subscription<tier4_rtc_msgs::msg::CooperateStatusArray_<std::allocator<void> >, std::allocator<void>, rclcpp::message_memory_strategy::MessageMemoryStrategy<tier4_rtc_msgs::msg::CooperateStatusArray_<std::allocator<void> >, std::allocator<void> > >::handle_message(std::shared_ptr<void>&, rclcpp::MessageInfo const&) () from /home/sig/autoware/install/rtc_manager_rviz_plugin/lib/librtc_manager_rviz_plugin.so
#5  0x00007ffff7097014 in ?? () from /opt/ros/galactic/lib/librclcpp.so
#6  0x00007ffff70978e4 in rclcpp::Executor::execute_subscription(std::shared_ptr<rclcpp::SubscriptionBase>) () from /opt/ros/galactic/lib/librclcpp.so
#7  0x00007ffff7098015 in rclcpp::Executor::execute_any_executable(rclcpp::AnyExecutable&) () from /opt/ros/galactic/lib/librclcpp.so
#8  0x00007ffff709c8fb in rclcpp::Executor::spin_some_impl(std::chrono::duration<long, std::ratio<1l, 1000000000l> >, bool) () from /opt/ros/galactic/lib/librclcpp.so
[... cut ...]
#23 0x00007ffff7444116 in QCoreApplication::exec() () from /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
#24 0x0000555555556a76 in ?? ()
#25 0x00007ffff6bad083 in __libc_start_main (main=0x5555555565f0, argc=3, argv=0x7fffffff38b8, init=<optimized out>, fini=<optimized out>, rtld_fini=<optimized out>, stack_end=0x7fffffff38a8) at ../csu/libc-start.c:308
#26 0x0000555555556b3e in ?? ()
(gdb) 
```

### Expected behavior

No crash

### Actual behavior

Rviz crashes

### Steps to reproduce

1. drive around with the planning simulator?

takayuki5168, 2022-10-27:
@taikitanaka3

taikitanaka3, 2022-10-27:
@VRichardJP 
Thank you I will see it.

taikitanaka3, 2022-10-27:
@VRichardJP 
This issue was caused by size check I will make PR for this fix
[image]

VRichardJP, 2022-10-27:
The program crashes here:
https://github.com/autowarefoundation/autoware.universe/blob/893a3bb725abebd83e00a0ebfdcffd616a5b24e5/common/rtc_manager_rviz_plugin/src/rtc_manager_panel.cpp#L400-L402

But it seems to be because `cnt` is too high:
```
(gdb) p cnt
$1 = 10
```

The `rtc_table_` has maximum 10 rows, but we can reach the 11th row because `cnt` can reach 10. I guess the line should be a `>= max_display_size`:
https://github.com/autowarefoundation/autoware.universe/blob/893a3bb725abebd83e00a0ebfdcffd616a5b24e5/common/rtc_manager_rviz_plugin/src/rtc_manager_panel.cpp#L331

taikitanaka3, 2022-10-27:
@VRichardJP 
can you review this PR?
https://github.com/autowarefoundation/autoware.universe/pull/2163

closed 2022-10-27 by taikitanaka3