autowarefoundation/autoware_universe #2167: Autoware ignores priority during avoidance manoeuvers

VRichardJP, 2022-10-27:
### Checklist

- [X] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [X] I've searched other issues and no duplicate issues were found.
- [X] I'm convinced that this is not my fault but a bug.

### Description

Consider the situation below: 
[image]

A vehicle is parked on the side of our lane. To avoid it, we have no choice but to cross the middle line. At the same time, a vehicle is coming from the opposite lane. Obviously, this vehicle has priority and autoware should wait until it has gone before engaging the maneuver.

As you can see in the simulation, autoware does not care about the vehicle coming from the opposite lane and wants to engage immediately.

### Expected behavior

Autoware should not leave its lane until the vehicle with priority is gone.

### Actual behavior

Autoware cross the middle lane and eventually crashes on the opposite vehicle.

### Steps to reproduce

1. set up scenario above in planning simulator

takayuki5168, 2022-10-31:
@satoshi-ota is dealing with this issue?

satoshi-ota, 2023-01-03:
I'm working on developping new feature for this issue. 

- https://github.com/autowarefoundation/autoware.universe/pull/2534
- https://github.com/autowarefoundation/autoware.universe/pull/2515