autowarefoundation/autoware_universe #2182: Trajectory planning for long vehicles not working properly

WJaworskiRobotec, 2022-10-31:
### Checklist

- [X] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [X] I've searched other issues and no duplicate issues were found.
- [X] I'm convinced that this is not my fault but a bug.

### Description

While testing of ITRI Bus on ITRI campus in AWSIM, I've discovered that the planner does not work properly for planning of turns for long vehicles. As can be noticed on the screenshot below, the bus is trying to "cut" the right turn by driving on the sidewalk. I know that there were seperate issues/discussions about expanding drivable area, but in this case we should think about different trajectory planning algorithm considering the dimensions of the vehicle. In reality, long vehicles like buses drive more to the outer part of the road, to avoid driving on the sidewalk.

[image]

The same behavior was observed in the planning_simulator, so it's not related to AWSIM:

[image]

Used vehicle model :
https://github.com/autowarefoundation/autoware.universe/issues

### Expected behavior

Bus properly driving on the entire map of ITRI campus. 

### Actual behavior

Bus gets stuck on the right turns.

### Steps to reproduce

1. Run planning simulator with ITRI map and cbus_vehicle
2. Set the goal poses to follow demo route 2 defined for ITRI project : [link](https://github.com/autowarefoundation/autoware-projects/issues/45#issuecomment-1274550309)

yukkysaito, 2022-11-09:
I think the current solution was to extend the drivable area to the opposite lane.
@satoshi-ota Can you explain the details?

satoshi-ota, 2022-11-09:
Hi @WJaworskiRobotec 
As @yukkysaito said, we have to expand the drivable for large vehicles, such as buses and trailers, to make turns at small intersections. In fact, the discussions is in progress [here](https://github.com/orgs/autowarefoundation/discussions/2851) about the feature to dynamically expand the drivable area, but this has not yet been implemented.

On the other hand, the feature to expand the drivable area **statically** already exists. The feature expands the width of the drivable area in module `behavior_path_planner` (e.g. [LaneFollowing](https://github.com/autowarefoundation/autoware.universe/blob/f1efcc2e163164ef40f4d1691aa6bc4e5cc9938d/planning/behavior_path_planner/src/scene_module/lane_following/lane_following_module.cpp#L139-L144)) by a specified amount, which can be changed with the following parameters.

https://github.com/autowarefoundation/autoware.universe/blob/f1efcc2e163164ef40f4d1691aa6bc4e5cc9938d/planning/behavior_path_planner/config/drivable_area_expansion.yaml#L1-L20

FYI:

- https://github.com/autowarefoundation/autoware.universe/pull/1973
- https://github.com/autowarefoundation/autoware.universe/issues/1854

WJaworskiRobotec, 2022-11-09:
Paths to cover as the first step are defined here : https://github.com/autowarefoundation/autoware-projects/issues/45

mehmetdogru, 2022-11-14:
@WJaworskiRobotec 

Here are my inputs/thoughts about the issue if [this temp solution](https://github.com/autowarefoundation/autoware.universe/issues/2182#issuecomment-1308229372) is used for this turn:

---

1) (out of context) First of all when I use [this vehicle model](https://github.com/autowarefoundation/cbus_vehicle_launch) and [itri_bus_sensor_kit](https://github.com/autowarefoundation/itri_bus_sensor_kit) I got the results as below. It seems like sensor kit-base_link-3D mesh are not properly aligned. If it is needed @kaancolak can give more detailed info about this and offer his help. For now I won't be using the 3D vehicle model to show the results.

[image]

- Idk if I alone encountered this but the 3D vehicle model shown in rviz seems to have an offset wrt base_link.
When I visualize the footprint of the vehicle `/planning/scenario_planning/lane_driving/motion_planning/surround_obstacle_checker/debug/footprint` calculated with given [vehicle parameters](https://github.com/autowarefoundation/cbus_vehicle_launch/blob/master/cbus_vehicle_description/config/vehicle_info.param.yaml) this is the result:

[image]

https://user-images.githubusercontent.com/48479081/201646903-f96c4e15-b631-4186-abbc-4650cfe36668.mp4

@WJaworskiRobotec If these are not my mistake I guess an issue should be created about it.

---

2) To add `road_border` tag  here is a way of how it can be done with [vector_map_builder](https://tools.tier4.jp/vector_map_builder_ll2/):

- Simply click on a linestring and choose `road_border` as type.
- Optionally you can select multiple linestrings by holding `SHIFT` key and clicking on the ones whose type you wanna change then type road_border.

https://user-images.githubusercontent.com/48479081/201650423-de235eb1-06b8-4ef4-acc6-52e9a8d50ebb.mp4

---

3) When we start the autoware with [drivable_area_expansion parameters](https://github.com/autowarefoundation/autoware.universe/blob/main/launch/tier4_planning_launch/config/scenario_planning/lane_driving/behavior_planning/behavior_path_planner/drivable_area_expansion.param.yaml) _(lets say 1.5 m for all -right&left-)_ still you wont see that it is not expanded on the road_border sides. 
The reason is that; when using the [expandLanelets function](https://github.com/autowarefoundation/autoware.universe/blob/b6f68a5b9f79fe09f38293e6622d92fd8bcbd73b/planning/behavior_path_planner/include/behavior_path_planner/utilities.hpp#L302-L304) you can specify the types you would like to skip. And it is `road_border` type as default. But only in avoidance_module it is given. So you have to add road_border type in [lane_following module](https://github.com/autowarefoundation/autoware.universe/blob/b6f68a5b9f79fe09f38293e6622d92fd8bcbd73b/planning/behavior_path_planner/src/scene_module/lane_following/lane_following_module.cpp#L139-L141) as an argument as well for this demo as it is done in [avoidance_module](https://github.com/autowarefoundation/autoware.universe/blob/b6f68a5b9f79fe09f38293e6622d92fd8bcbd73b/planning/behavior_path_planner/src/scene_module/avoidance/avoidance_module.cpp#L1779-L1781).

@maxime-clem Is this done intentionally? If not shall we create a PR where all the modules use road_border arg as default to skip the expansion?

---

4) Then we will have and expanded lanelet only on sides we want it to be:

https://user-images.githubusercontent.com/48479081/201657417-6be2bff3-a4a2-43e1-9252-6256ed324837.mp4

In the video you will see that side of the vehicle is outside of drivable_area, yet it doesn't stop. It is because in obstacle_avoidance_module only the corners are checked if they are out or not. This issue is known and worked on in #1999 and will be merged soon hopefully. 

However it possible to handle the turns better since there is lots of space available yet we are not able to turn safely. It can be accomplished by enabling the parameter [kinematics.optimization_center_offset](https://github.com/autowarefoundation/autoware.universe/blob/b6f68a5b9f79fe09f38293e6622d92fd8bcbd73b/launch/tier4_planning_launch/config/scenario_planning/lane_driving/motion_planning/obstacle_avoidance_planner/obstacle_avoidance_planner.param.yaml#L72-L76) and assigning `0` to it. Which it will make it possible that base_link will be optimization center and yields more realistic optimization results. [Here](https://autowarefoundation.github.io/autoware.universe/main/planning/obstacle_avoidance_planner/#getmodelpredictivetrajectory) is more info about it. To me for long vehicles such as buses optimization center should be base_link. Idk also as default if we should change it or not @mitsudome-r @maxime-clem? On isuzu bus but we use base_link as optimization center. 

Results after changing the optimization center:

https://user-images.githubusercontent.com/48479081/201664436-b1ab0fee-c341-4a1e-86e7-3a5015e4cf47.mp4

takayuki5168, 2022-11-15:
@mehmetdogru

> To me for long vehicles such as buses, optimization center should be base_link

I agree with this idea.
The default tuning of kinematics.optimization_center_offset is for a normal car. Therefore, I think it's good to set 0 for buses.
If you have further question about tuning of `obstacle_avoidance_planner` package, feel free to ask me.

maxime-clem, 2022-11-15:
> @maxime-clem Is this done intentionally? If not shall we create a PR where all the modules use road_border arg as default to skip the expansion?

You are right and this is something I overlooked in my previous PR. I will prepare a new PR to fix this.

closed 2023-01-12 by maxime-clem