autowarefoundation/autoware_universe #2186: Enable OpenCL Backend for TVM

angry-crab, 2022-11-01:
### Checklist

- [X] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [X] I've searched other issues and no duplicate issues were found.
- [X] I've agreed with the maintainers that I can plan this task.

### Description

Currently, we don't have a working opencl pipeline with tvm. As discussed in #1898, vulkan backend cannot be used because of an unknown accuracy issue and llvm does not provide enough performance for deep learning inference. Either cuda or opencl has to be enabled. 

### Purpose

We need to enable opencl backend so that tvm packages could be actually used. 

### Possible approaches

- [ ] Compile the models for opencl. In theory it is only a matter of adding opencl here: https://github.com/autowarefoundation/modelzoo/blob/9529c71e73df637448f6fceb203fa793f3a3b064/scripts/tvm_cli/validation/tvm_cli_test.py#L19, then it gets picked up by the CI.
- [ ] Enable opencl in the runtime package by modifying this file: https://github.com/autowarefoundation/tvm_vendor/blob/main/config.cmake.gpu.patch, and then waiting for the package to be tagged and released.
- [ ] Making sure the correct opencl dependencies are installed in the Autoware docker image / that they can be installed easily for source installs.

### Definition of done

Being able to run tvm packages using opencl.

angry-crab, 2022-11-01:
Related TVM issues/PR:
#1988 #2185 
#908 #1898

angry-crab, 2022-11-02:
A simple verification of compiled opencl models indicates accuracy issues exist with output channel `heatmap` and `vel` of `pts_backbone_neck_head_centerpoint.onnx`.
```
# heatmap
(1, 3, 560, 560)
(1, 3, 560, 560)
[0.0001049  0.00010109 0.0001297  ... 0.00014496 0.000103   0.00011253]
1375

# vel
(1, 2, 560, 560)
(1, 2, 560, 560)
[0.00010395 0.0001142  0.00010109 0.00011301]
4
```

ambroise-arm, 2022-11-02:
Are you still using the same setup as in https://github.com/autowarefoundation/autoware.universe/pull/1898#issuecomment-1293085900? Using it for the backbone outputs, with `np.ones((40000,32,9), dtype=np.float32)` as an input, I don't see any difference delta above 0.0001 with opencl (or vulkan). Changing the threshold to 0.00001 there are differences, but then I also see differences with an llvm backend.

angry-crab, 2022-11-07:
> Are you still using the same setup as in [#1898 (comment)](https://github.com/autowarefoundation/autoware.universe/pull/1898#issuecomment-1293085900)? Using it for the backbone outputs, with `np.ones((40000,32,9), dtype=np.float32)` as an input, I don't see any difference delta above 0.0001 with opencl (or vulkan). Changing the threshold to 0.00001 there are differences, but then I also see differences with an llvm backend.

For backbone network, the input shape has to be changed. It's (1, 32, 560, 560) I think. 
[image]

angry-crab, 2022-11-07:
[add opencl backend to ci](https://github.com/autowarefoundation/modelzoo/pull/79)

ambroise-arm, 2022-11-07:
@angry-crab Yes, sorry I had copied and pasted the wrong line here, but I was using the correct input shape in my script.
I confirm that I don't see deltas when using `np.ones`, but I do when using `np.zeros`. Although only on `vel`. Both vulkan and opencl give me:
```
[0.0001049  0.00011146 0.00014421 0.00010443 0.0001049  0.0001266 0.00011784 0.00010717 0.00010872 0.00011826 0.00012851 0.00014114 0.00012398]
13
```
And llvm gives me:
```
[0.00010467]
1
```

Is there a reason you are using ones as input to the encoder and zeros as input to the backbone?

angry-crab, 2022-11-08:
> Is there a reason you are using ones as input to the encoder and zeros as input to the backbone?

Not really. It seems the accuracy is hardware dependent. Let's see how much it affects the detection.

angry-crab, 2022-11-09:
@ambroise-arm
As for the third mark, since `tvm_vendor` depends on `opencl`, the image should be good if correct version of `tvm_vendor` is installed, isn't it?

ambroise-arm, 2022-11-09:
Oh, absolutely! I hadn't thought about that.

angry-crab, 2022-11-10:
[image]
Did a quick test on `opencl`, it seems to work.

ambroise-arm, 2022-11-10:
Nice! Was that locally or inside of a docker image?
Because I think there are `.icd` files that will be missing inside a docker image. `mesa-opencl-icd` would need to be installed when using an AMD GPU, and `echo "libnvidia-opencl.so.1" > /etc/OpenCL/vendors/nvidia.icd` would need to be issued for an NVIDIA GPU. Probably the same way we do here: https://github.com/autowarefoundation/autoware/blob/5fba5a1e35921470d02711f19bec19b29ab08139/docker/autoware-universe/Dockerfile#L45 for Vulkan.

angry-crab, 2022-11-10:
> Nice! Was that locally or inside of a docker image? Because I think there are `.icd` files that will be missing inside a docker image. `mesa-opencl-icd` would need to be installed when using an AMD [...]

Yes! I was using a docker image and `echo "libnvidia-opencl.so.1" > /etc/OpenCL/vendors/nvidia.icd` is required.

angry-crab, 2022-11-11:
It seems that `tvm_vendor` will be released on November 21, 2022 (Chicago) for humble. 
[Humble Release](https://discourse.ros.org/t/preparing-for-humble-sync-and-patch-release-2022-11-21/28234)

angry-crab, 2022-11-25:
Humble packages are still being held. Not sure why.

angry-crab, 2022-12-08:
Humble package is merged! [tvm_vendor](https://github.com/ros/rosdistro/pull/35276)

angry-crab, 2022-12-23:
The only issue left is that tvm_vendor in the existing docker images are not update to date.

angry-crab, 2022-12-27:
@ambroise-arm 
As a follow-up issue of GPU backend, shall we enable cuda as well? According to a performance comparison survey. opencl is 13% ~ 67% slower than cuda.

ambroise-arm, 2023-01-02:
There is an issue with doing that in practice. The way the TVM runtime works, it expects all the dependencies it was built with to be available when used. So if we compile [tvm_vendor](https://github.com/autowarefoundation/tvm_vendor) with `set(USE_CUDA ON)`, the user will need to have the nvidia cuda library available in order for the inference to work, even if the user just wants to do non-cuda inference.
At least that is the way it was about a year ago, last time I looked at it.

It is not a problem for vulkan and opencl backends because there are mesa packages available that provide the corresponding libraries.

angry-crab, 2023-01-10:
> It is not a problem for vulkan and opencl backends because there are mesa packages available that provide the corresponding libraries.

I see. But if cuda is installed then we can safely add `set(USE_CUDA ON)` and compile tvm runtime. I think maybe we should provide a different cmake setting for cuda.

angry-crab, 2023-01-10:
OpenCL is correctly setup in the latest cuda docker image. We can close this issue.

closed 2023-01-12 by angry-crab