autowarefoundation/autoware_universe #2200: Can't see bounding box in rviz.

nainaigetuide, 2022-11-02:
### Checklist

- [X] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [X] I've searched other issues and no duplicate issues were found.
- [X] I'm convinced that this is not my fault but a bug.

### Description

I use the following command.
ros2 launch autoware_launch logging_simulator.launch.xml map_path:=$HOME/work/rosbag_map/universe/sample-map-rosbag vehicle_model:=sample_vehicle sensor_model:=sample_sensor_kit | grep centerpoint

I get the log as below.

[INFO] [lidar_centerpoint_node-27]: process started with pid [586904]
[lidar_centerpoint_node-27] Loading from /home/autoware/install/lidar_centerpoint/share/lidar_centerpoint/data/pts_voxel_encoder_centerpoint_tiny.engine
[lidar_centerpoint_node-27] Loading from /home/autoware/install/lidar_centerpoint/share/lidar_centerpoint/data/pts_backbone_neck_head_centerpoint_tiny.engine
[lidar_centerpoint_node-27] [WARN] [1667293552.178222006] [lidar_centerpoint]: Could not find a connection between 'base_link' and 'map' because they are not part of the same tree. Tf has two or more unconnected trees.
[lidar_centerpoint_node-27] [WARN] [1667293552.178362394] [lidar_centerpoint]: Fail to preprocess and skip to detect.
[lidar_centerpoint_node-27] [WARN] [1667293552.688393673] [lidar_centerpoint]: Could not find a connection between 'base_link' and 'map' because they are not part of the same tree.Tf has two or more unconnected trees

In the GPU monitoring, I can see the relevant process information and the normal use of the GPU.
[image]

According to the above information, you can see that the engine has been generated and the GPU is in normal use. Do you know why I can't see the blue rectangle around objects like cars on other lanes?
[image]

### Expected behavior

The bounding box can be seen in rviz.

### Actual behavior

Can't see bounding box in rviz.

### Steps to reproduce

1. ros2 launch autoware_launch logging_simulator.launch.xml map_path:=$HOME/work/rosbag_map/universe/sample-map-rosbag vehicle_model:=sample_vehicle sensor_model:=sample_sensor_kit | grep centerpoint
2. ros2 bag play /home/inspur/Downloads/sample-rosbag

### Versions

-OS: Ubuntu 20.04
-ROS: Galactic

### Possible causes

none

### Additional context

none

maxime-clem, 2022-11-04:
I have not been able to reproduce the issue with the latest version of Autoware following the instructions from https://autowarefoundation.github.io/autoware-documentation/main/tutorials/ad-hoc-simulation/rosbag-replay-simulation/.
Did you make any modification to Autoware or to the Rviz configuration ?
Also, can you please share the full log file found in `$HOME/.ros/log/DATE_TIME/launch.log` ?

nainaigetuide, 2022-11-04:
@maxime-clem 
[launch.log](https://github.com/autowarefoundation/autoware.universe/files/9934952/launch.log)
 The full log file is shown in the attachment.

I mainly use the https://github.com/tier4/CARET_report tool, but this tool is mainly used for tracing and should not affect perception module in theory.

maxime-clem, 2022-11-04:
Thank you for sharing the full log file. I cannot see any issue from the log file.
Are you able to see some messages when running the following command ?
```
ros2 topic echo /perception/object_recognition/objects
```
If you see some messages being published, then maybe the issue comes from rviz. Can you make sure the visualization of the `/perception/object_recognition/objects` topic is correct ?
[image]

nainaigetuide, 2022-11-04:
@maxime-clem 
Thank you, using the command the result is as follow.
`ros2 topic echo /perception/object_recognition/objects`
`The message type 'autoware_auto_perception_msgs/msg/PredictedObjects' is invalid`

maxime-clem, 2022-11-04:
> @maxime-clem Thank you, using the command the result is as follow. `ros2 topic echo /perception/object_recognition/objects` `The message type 'autoware_auto_perception_msgs/msg/PredictedObjects' is invalid`

You need to run `source install/setup.sh` in the autoware folder and then it should work.

nainaigetuide, 2022-11-04:
@maxime-clem 
No message is published.
[image]

nainaigetuide, 2022-11-07:
@maxime-clem 
I further analyzed the .log and found that the node object_lanelet_filter_node is not working. The relevant logs are as follows.

`[object_lanelet_filter_node-31] /home/inspur/autoware/install/detected_object_validation/lib/detected_object_validation/object_lanelet_filter_node: symbol lookup error: /home/inspur/autoware/install/detected_object_validation/lib/libobject_lanelet_filter.so: undefined symbol: _ZNK5utils17FilterTargetLabel8isTargetEh`

`$ c++filt _ZNK5utils17FilterTargetLabel8isTargetEh`
`utils::FilterTargetLabel::isTarget(unsigned char) const`

Recompiling the detected_object_validation package does not solve the above problem.

Do you have a better suggestion?

maxime-clem, 2022-11-07:
@nainaigetuide 
> Do you have a better suggestion?

To completely recompile the `detected_object_validation` package, it may be necessary to first remove the corresponding build and install folders in your workspace:
```
rm build/detected_object_validation install/detected_object_validation -rf
```
Can you try that and see if it fixes the issue ?

nainaigetuide, 2022-11-07:
@maxime-clem  
Thank you. Solved in the following way.
https://github.com/autowarefoundation/autoware.universe/commit/cf91d1a1425e8ee0682af91add418936788932aa
The specific reason is that my version is a little behind the current latest version.

closed 2022-11-07 by nainaigetuide