autowarefoundation/autoware_universe #2208: behavior_path_planner creates shift point outside drivable area

beyzanurkaya, 2022-11-03:
### Checklist

- [X] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [X] I've searched other issues and no duplicate issues were found.
- [X] I'm convinced that this is not my fault but a bug.

### Description

`behavior_path_planner` creates a shift point outside drivable area while avoiding the obstacle.

### Expected behavior

The shift points behavior_path_planner creates must always be inside drivable area.

### Actual behavior

In some cases the shift point comes out of the drivable area.

### Steps to reproduce

1. Download isuzu_vehicle_launch in [here]( https://drive.google.com/drive/folders/1bkxamrkKsbP3J5339GVf5yKhJL0iii4l)
2. Download [test.zip](https://github.com/autowarefoundation/autoware.universe/files/9926299/test.zip)
3. Change `publish_debug_marker` parameter to true in avoidance.param.yaml 
4. run `ros2 launch scenario_test_runner scenario_test_runner.launch.py sensor_model:=sample_sensor_kit vehicle_model:=isuzu_vehicle \ scenario:=/scenario/path/test.yaml \  architecture_type:=awf/universe launch_rviz:=false launch_autoware:=true`

maxime-clem, 2022-11-04:
Can you try this PR: https://github.com/autowarefoundation/autoware.universe/pull/2196 ? I think it will fix this issue.

mehmetdogru, 2022-11-04:
@beyzanurkaya 

Could you please add some screenshots or videos explains the issue?

beyzanurkaya, 2022-11-08:
> Can you try this PR: #2196 ? I think it will fix this issue.

I already created a PR for this issue actually.

mehmetdogru, 2022-12-05:
I will close the issue since I was not able to reproduce it anymore.

closed 2022-12-05 by mehmetdogru