autowarefoundation/autoware_universe #2211: ‘l2_pseudo_jerk_smoother’ takes  long  time to start and speed up at low speed by default param

shulanbushangshu, 2022-11-03:
### Checklist

- [X] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [X] I've searched other issues and no duplicate issues were found.
- [X] I'm convinced that this is not my fault but a bug.

### Description

The "l2_pseudo_jerk_smoother" takes too long time to start by the default parameters in the "motion_velocity_smoother.param.yaml". When the vehicle is turning, it cannot accelerate normally after the speed is reduced, and it takes a long time to recover the normal speed.

https://user-images.githubusercontent.com/102840938/199663870-ba2b7e81-3827-4bdb-9e03-5d3284d86d3e.mp4

### Expected behavior

The ‘l2_pseudo_jerk_smoother’ shall take normal time to start and shall accelerate normally when the speed is reduced to low value.

### Actual behavior

The ‘l2_pseudo_jerk_smoother’ takes too long time to start.When the vehicle is turning, it cannot accelerate normally after the speed is reduced, and it takes a long time to recover the normal speed.

### Steps to reproduce

1.Chose the "L2" in the "scenario_planning.launch.xml"
2.Start autoware planning simulator;
3. Set initial position around the bend

### Versions

1.OS: Ubuntun 20.04
2.ROS2: Galactic
3.Autoware: Autoware.universe:main

### Additional context

 There is no slow acceleration for other smoothers ("JerkFiltered","Analytical","Linf")  by the default parameters  in the "motion_velocity_smoother.param.yaml"

takayuki5168, 2022-11-12:
cc @rej55 @purewater0901 @TakaHoribe

shulanbushangshu, 2022-11-15:
I have test the code.I find the value of cost about jerk is large.
(1)When I change  the value of pseudo_jerk_weight in the "L2.param.yaml",as below:

[image]

The car can take normal time to start;
(2)When I change the code of cost about jerk,as below:
[image]

I do not square the smooth_weight in the calculation
The car can take normal time to start.

closed 2022-11-21 by shulanbushangshu