autowarefoundation/autoware_universe #2213: Obstacle on the path reported when the detected object is not directly on the path

WJaworskiRobotec, 2022-11-03:
### Checklist

- [X] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [X] I've searched other issues and no duplicate issues were found.
- [X] I'm convinced that this is not my fault but a bug.

### Description

While testing current Autoware in AWSIM for ITRI BusODD scenarios, the following problem was discovered :

The vehicle stops because the obstacle on the path is reported, but when detected objects are visualized, it seems that the objects are not really blocking the path.

- the only detected object is far from the vehicle, and nnot on the path
[image]

- detected object is on the side of the vehicle, and is not blocking the movement

[image]
[image]
[image]

### Expected behavior

 The vehicle should not stop if there are no objects detected on the path

### Actual behavior

The vehicle stops with no objects on the path

### Steps to reproduce

Run AWSIM binary with Autoware - description to be updated once binary is publicly shared

maxime-clem, 2022-11-04:
An "obstacle on the path" can also be caused by an obstacle _point_.
Have you checked the pointcloud in rviz ? I am afraid the points from the tree leaves are what is causing your issue.
Here is the related open discussion: https://github.com/orgs/autowarefoundation/discussions/2981

shmpwk, 2022-11-04:
@WJaworskiRobotec 
This problem may comes from [this PR](https://github.com/autowarefoundation/autoware.universe/pull/1899/files).
Could you change these parameters https://github.com/autowarefoundation/autoware.universe/blob/main/launch/tier4_perception_launch/config/obstacle_segmentation/ground_segmentation/ground_segmentation.param.yaml#L14-L15 ?

before:
```
max_z: 10.7 
min_z: -8.7
```

After
```
max_z: 4.5 # in case 50m continuous slope at 5deg
min_z: -4.5# in case 50m continuous slope at 5deg
```

shmpwk, 2022-11-04:
cc. @badai-nguyen

WJaworskiRobotec, 2022-11-05:
@maxime-clem so what is the input to the planner from sensing/perception modules ? I thought it's only list of detection objects, but from your comments I understand that it's a pointcloud after some segmentation/cropping as well, right ?

maxime-clem, 2022-11-06:
@WJaworskiRobotec You are right that the `obstacle_avoidance_planner` only uses the list of detected objects but some other planning modules uses the obstacle pointcloud (most notably the `obstacle_stop_planner`).
[This page](https://autowarefoundation.github.io/autoware-documentation/main/design/autoware-architecture/planning/) gives a good overview of which planning nodes uses which inputs.

yukkysaito, 2022-11-09:
cc @takayuki5168

xmfcx, 2022-11-09:
https://www.thinkautonomous.ai/blog/occupancy-networks/

shmpwk, 2022-11-10:
Reflected https://github.com/autowarefoundation/autoware.universe/issues/2213#issuecomment-1303143710  by
https://github.com/tier4/autoware_launch/pull/543
https://github.com/autowarefoundation/autoware.universe/pull/2260

lchojnack, 2022-11-21:
I've found the root cause of stopping the vehicle presented in the issue, and also in the discussion https://github.com/orgs/autowarefoundation/discussions/2981. I've noticed that in these two situations there are tree branches and tree leaves on the EGO path. The tree branches are higher than the vehicle's height and do not cause danger or reason to stop the vehicle. The node which is responsible for detecting points that are on the path is `obstacle_stop_planner`. The `obstacle_stop_planner` is checking if any point from a point cloud is within the polygon trajectory, and decides to stop the vehicle. Checking points only within a polygon (2d information) and not taking into account the vehicle height could cause stopping the vehicle when an obstacle is above and below the vehicle. There is an example of detecting tree branches above the vehicle in the picture below.
[image]

My proposition is to extend checking if the point is within **a polyhedron** (3d information) instead of if is within a polygon (2d information). The idea is presented in the picture below.
[image]

xmfcx, 2023-02-21:
@WJaworskiRobotec is this issue now resolved with https://github.com/autowarefoundation/autoware.universe/pull/2296 merged?

WJaworskiRobotec, 2023-02-21:
@xmfcx , Yes, this problem doesn't happen anymore, at least in the situation that was reported here. Issue can be closed.

closed 2023-02-21 by xmfcx