autowarefoundation/autoware_universe #2271: Port xsens drivers from autoware_auto to universe

reinierkof, 2022-11-11:
### Checklist

- [X] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [X] I've searched other issues and no duplicate issues were found.
- [X] I've agreed with the maintainers that I can plan this task.

### Description

In autoware auto there are drivers for xsens sensors, these are used for IMU and GPS. They can be found here:
https://autowarefoundation.gitlab.io/autoware.auto/AutowareAuto/namespaceautoware_1_1drivers_1_1xsens__driver.html

However these drivers are not yet ported to autoware universe.

I want to port these drivers to Universe such that the xsens sensors are compatible with autoware universe.

### Purpose

For xsens sensors to be compatible with this version of autoware

### Possible approaches

- Take the template for drivers already in autoware universe
- Take the drivers written in autoware_auto
- Convert the variables written in autoware_auto to the right spots in autoware universe

### Definition of done

Xsens sensor to be compatible with autoware universe.

VRichardJP, 2022-11-11:
I am curious about this, because I have been using Xsens MTi devices with autoware universe for a while, and the official driver works just fine: https://github.com/bluespace-ai/bluespace_ai_xsens_ros_mti_driver.

Is there any special need only convered by autoware.auto's implementation?

reinierkof, 2022-11-11:
@VRichardJP Oh, i came across these drivers and thought they wouldn't instantly work with autoware.
Did you make any changes to your autoware setup or to the blue space setup? Or did it work straight out of the box?

And is it then an idea to add it to the documentation of sensing?

VRichardJP, 2022-11-11:
Yes, it is really easy to use. My vehicle sensor kit `imu.launch.xml` file simply includes the `xsens_mti_node.launch.py` with my custom yaml parameter file. The node can output pretty much everything the Xsens device can measure (IMU, gnss, magnetic field, etc).

The only modification I have ever made to the driver is to add RTCM corrections data forwarding to the device to use RTK-GNSS. But for this I had just followed the official documentation (e.g. https://base.xsens.com/s/article/How-to-receive-RTCM-correction-messages-for-MTi-680-G?language=en_US)

closed 2022-11-15 by reinierkof